olikraus/u8g2 #2591: ST7588

bbd666, 2025-02-12:
line 172 of u8x8_d_st7588.c

  /* pixel_width = */ 128,
  /* pixel_height = */ 64

but according to ST7588 datasheet
it's 181 x 132 Dot Matrix LCD

olikraus, 2025-02-12:
I think the ST7588 81 x 132 memory.
Anyhow, there are two distinct things:

1) Display Controller, which has a certain amount of memory to store the LCD/OLED content
2) The LCD (or OLED) itself, which has a certain number of pixels.

However the display controller could also support larger displays.

Lets take the SH1106 OLED controller: It supports OLEDs with up to 132x64 pixel.
However it is bundled in 99% of all cases with a 128x64 screen. 4x64 bits are wasted, but I guess this is still cheaper than doing an additional development for an 128x64 controller.

In other words: In most cases the display controller will support bigger screens than actually supported by the display.

In the above code the pixel_width and height does NOT refer to the controller, but instead is the pixel size of the display.

One more hint: You should always receive two datasheets: One for the display and another for the controller.
For example here: https://www.buydisplay.com/2-8-inch-white-192x96-graphic-lcd-display-module-st75256-for-arduino
First datasheet is for the display itself (192x96 pixel), second is for the display controller (256x128 pixel).

bbd666, 2025-02-13:
ok thks.

bbd666, 2025-02-13:
something else that's not clear to me: 
in your list I see ST7588 and KS0108.
I read somewhere that ST7588 is member of KS0108 family.
Then KS0108 controller should also drive a ST7588 does'nt it ?

olikraus, 2025-02-17:
To my knowledge these two display controllers are very different. The have a different command set and a different resolution (as far as I can remember)