olikraus/u8g2 #2592: default_x_offset on sh1106

Coding-Kiwi, 2025-02-13:
Hi,

thanks for the awesome library.
I got one of these https://www.az-delivery.de/products/1-3zoll-i2c-oled-display

and I am using the `U8G2_SH1106_128X64_NONAME_F_HW_I2C` constructor.

```cpp
U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(
    /* rotation=*/U8G2_R0,
    /* reset=*/U8X8_PIN_NONE,
    /* clock=*/DISPLAY_CLOCK,
    /* data=*/DISPLAY_DATA);

u8g2.begin();
u8g2.clearBuffer();
u8g2.drawFrame(0, 0, 128, 64);
u8g2.sendBuffer();
```

sadly the left two pixels are cut off / behave weirdly

width 128 draws one pixel at x 0 y 10-ish? and a double line at x 1 and x 2
width 127 draws a full line at x 0 , a blank at x 1 and a full at x 2

the only way I could fix this is by setting `default_x_offset` of `u8x8_sh1106_128x64_noname_display_info` from 2 to 0 in the `u8x8_d_ssd1306_128x64_noname.c` file

maybe you can give me a hint.. is there any way to do this in my code instead of modifying stuff inside the library folder?

Thanks!

olikraus, 2025-02-17:
Maybe try a SSD1306 constructor instead.

Coding-Kiwi, 2025-02-18:
Thank you, I am stupid, solved.

closed 2025-02-18 by Coding-Kiwi