olikraus/u8g2 #2595: Trouble With SSD1322 OLED 4-Wire Hardware SPI Using ESP32 HSPI Pins

zenith2024, 2025-02-16:
Can't seem to get anywhere with this problem, i've scoured the issues page here like [#1364](https://github.com/olikraus/u8g2/issues/1364) and [#1364](https://github.com/olikraus/u8g2/issues/1364) but am still stuck in a rut. 

I've tried the default HSPI pins to no avail. When using software SPI it works perfectly, although **it only displays at 2FPS**(even in the u8g2 example projects)... which is exactly the reason why i was diving into hardware SPI in the first place

Code Below:

```
#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

#define SPI_MOSI_PIN 13   // Blue
#define SPI_SCK_PIN  14   // Green
#define SPI_CS_PIN   15   // Purple

#define SPI_RES_PIN  16   // Yellow
#define SPI_DC_PIN   4    // White
#define SPI_MISO_PIN 19   // ???

//U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, SPI_CS_PIN, SPI_DC_PIN, SPI_RES_PIN);
U8G2_SSD1322_NHD_256X64_F_4W_SW_SPI u8g2(U8G2_R0, /*SCK=*/SPI_SCK_PIN , /*MOSI=*/SPI_MOSI_PIN, /*CS=*/SPI_CS_PIN, /*DC=*/SPI_DC_PIN , SPI_RES_PIN);

// rest of my code
```

This is the pinout of my SSD1322 display;

[image]

Any help would be amazing, Cheers!

closed 2025-02-16 by zenith2024

olikraus, 2025-02-17:
Is this topic solved?

danielskowronski, 2025-04-04:
I have the same problem with Arduino ESP32 2.0.18 and apparently ESP32S

When using `U8G2_SSD1322_NHD_256X64_F_4W_SW_SPI`, the refresh rate is very poor.

Semi workaround for was to use `U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI` without declaring pins in code but physically use VSPI pins (18 for SCLK and 23 for SDI) **and** to use WiFi. To me, it looks like issue caused by incorrect labels in SDK (https://github.com/espressif/arduino-esp32/issues/10079) and the issue https://github.com/espressif/arduino-esp32/issues/5923 seems to exactly the same as our one with u8g2. 

I was not able to replicate what WiFi init does by experimenting with `SPI = SPIClass(...)` and  `SPI.begin(18,-1,23,-1)` on both HSPI (14,13) and VSPI (18,23) pins.

Issues likely related to this mess:

- https://github.com/olikraus/u8g2/issues/2614
- https://github.com/olikraus/u8g2/issues/1061

---

If I get this (and code) right (and I may be mistaken, it's nearly 2AM), software emulation of SPI (`SW`) with full frame buffer (`F`) is so heavy that it offers max 2 FPS even on ESP32. What we want to use with `F` is `HW` - either on HSPI or VSPI since *both* those buses are hardware SPI (there are also two other SPIs for internal flash). ESP32-S2 seems to have nice addition of having one of hardware SPI buses (SPI3, which is labelled VSPI on other boards) which can use any pin (sort of hardware accelerated "software" SPI). 

Furthermore, due to the complexity of both Arduino SDK for ESP32 and U8x8 there seems to be issue with properly initializing VSPI on HW mode and only `WiFi.begin()` redefines something.

olikraus, 2025-04-11:
well... I tried to support ESP32, but it is not an official Arduino Port and indeed it differs a lot...