olikraus/u8g2 #2598: st7302 high performance mode flashing

yueduz, 2025-02-20:
Has anyone tried the high-performance mode? When I use the x38 command to enable high-performance mode, the screen keeps flickering.

olikraus, 2025-02-20:
I am confused. Isn't the high power mode activated already? 
https://github.com/olikraus/u8g2/blob/5f969586c7659fc236fb650c46c8c2f068abe29f/csrc/u8x8_d_st7302.c#L315

yueduz, 2025-02-21:
https://github.com/olikraus/u8g2/blob/5f969586c7659fc236fb650c46c8c2f068abe29f/csrc/u8x8_d_st7302.c#L342C1

It is set to low-power mode in line 342

olikraus, 2025-02-22:
oh, ok I see. Hmm maybe then I should remove the cmd in line 315.

> When I use the x38 command to enable high-performance mode, the screen keeps flickering.

Maybe I encountered the same problem and switched back to low power mode (I don't remember).
Often the commands depend on each other and it is difficult to find a combination of all the commands, which leads
to an acceptable result.

In detail:

- In high power mode, try a different frame rate (0xB2 cmd). In u8g2 this is set to 1 (=32Hz). Check out 0 (=16Hz). Maybe the flicker is gone.
- Look out for other commands, which do influence the timing. Maybe 0xB8 could be such a value which needs to be configured in a different way for high power mode.

yueduz, 2025-02-23:
I tried many combinations, none of them worked. But I found that 32Hz flickering is more serious than 16Hz. The video is 16hz case.

https://github.com/user-attachments/assets/467e1d5d-34ca-4865-937b-35efa64b4ebe

yueduz, 2025-02-23:
I tried using 'U8X8_C (0x12),//Partial On' to avoid flickering. When the screen is still, it does not flicker, but it has an extra strong line that should not appear.

[image]

olikraus, 2025-02-24:
hmm.. not sure how I can help here... Sometimes a display just doesn't support all features of the display controller.