olikraus/u8g2 #2602: Raspberry PI drawUTF8Button, method not found

ironlungx, 2025-03-02:
TLDR: `u8g2.drawButtonUTF8()` doesn't work on raspberry pi zero 2 w. Possible issue is that `port/U8g2lib.h` is outdated from the main `cppsrc/`

Helllllo, I was porting my library to the raspberry pi (it depends on u8g2). In one of the functions I use `u8g2.drawButtonUTF8(...)`. It works perfectly normally on the microcontrollers I've tested it with, but it can't find that method on the pi zero 2w.
```
u8g2->drawButtonUTF8(42, 50, U8G2_BTN_INV | U8G2_BTN_SHADOW2 | U8G2_BTN_HCENTER | U8G2_BTN_BW1, 0, 2, 2, "Yes");
```

u8g2 is a pointer because of the way the library is structured. 

```bash
$ make -j $(nproc)
lib/pixel-view/src/pixelView.cpp:86:15: error: ‘class U8G2’ has no member named ‘drawButtonUTF8’; did you mean ‘drawExtUTF8’?
   86 |         u8g2->drawButtonUTF8(42, 50, U8G2_BTN_INV | U8G2_BTN_SHADOW2 | U8G2_BTN_HCENTER | U8G2_BTN_BW1, 0, 2, 2, "Yes");
```

This is my Makefile
```make
PI=pi@pi.local:~/

CC=arm-linux-gnueabihf-gcc
CXX=arm-linux-gnueabihf-g++

TARGET=main
LIBDIR=./lib

U8G2_PORTDIR=$(LIBDIR)/u8g2/sys/arm-linux/port
U8G2_DRIVERS=$(LIBDIR)/u8g2/sys/arm-linux/drivers
U8G2_CSRC=$(LIBDIR)/u8g2/csrc

PV_DIR=$(LIBDIR)/pixel-view/src

# Add PixelView directory to include paths
IDIR= -I $(U8G2_DRIVERS) -I $(U8G2_DRIVERS) -I $(U8G2_PORTDIR) -I $(U8G2_CSRC) -I $(PV_DIR)
CSRCDIR=$(U8G2_CSRC)
CXXSRCDIR=$(U8G2_PORTDIR)

OBJDIR=./obj
OUTDIR=./bin
LDIR= -L ./lib
LIBS= -lm

CFLAGS= $(IDIR) -W -Wall -D __ARM_LINUX__

OBJ+=$(OBJDIR)/main.cpp.o\
	$(U8G2_PORTDIR)/u8g2port.o\
	$(U8G2_DRIVERS)/gpio.o\
	$(U8G2_DRIVERS)/spi.o\
	$(U8G2_DRIVERS)/i2c.o\
	$(OBJDIR)/Print.cpp.o\
	$(OBJDIR)/U8x8lib.cpp.o\
	$(OBJDIR)/U8g2lib.cpp.o\
	$(OBJDIR)/pixelView.cpp.o\

OBJ+=$(patsubst $(CSRCDIR)/%.c,$(OBJDIR)/%.o, $(wildcard $(CSRCDIR)/*.c))

all: directories $(TARGET) 

directories:
	@mkdir -p $(OBJDIR)
	@mkdir -p $(OUTDIR)

$(TARGET):$(OBJ)
	@echo Generating $(TARGET) ...
	@$(CXX) -o $(OUTDIR)/$@ $(OBJ) $(LDIR) $(LIBS)

$(OBJDIR)/%.cpp.o: $(CXXSRCDIR)/%.cpp
	$(CXX) -c -o $@ $< $(CFLAGS) $(LDIR) $(LIBS)

$(OBJDIR)/%.o: $(CSRCDIR)/%.c
	$(CC) -c -o $@ $< $(CFLAGS) $(LDIR) $(LIBS)

$(OBJDIR)/main.cpp.o: ./main.cpp
	$(CXX) -c -o $@ $< $(CFLAGS) $(LDIR) $(LIBS)

# Add rule to compile pixelView.cpp
$(OBJDIR)/pixelView.cpp.o: $(PV_DIR)/pixelView.cpp
	$(CXX) -c -o $@ $< $(CFLAGS) $(LDIR) $(LIBS)

clean:
	@echo RM -rf $(OBJDIR)/
	@rm -rf $(OBJ)
	@rm -rf $(OBJDIR)
	
	@echo RM -rf $(OUTDIR)/
	@rm -rf $(OUTDIR)

upload:
	@scp $(OUTDIR)/$(TARGET) $(PI)
```

The only thing I could conclude is that `port/U8g2lib.h` is outdated.

ironlungx, 2025-03-02:
This is the most basic example I could test with

```cpp
#include "U8g2lib.h"

#define I2C_BUS 1
#define I2C_ADDRESS 0x3c * 2

U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/U8X8_PIN_NONE);

int main() {
	u8g2.initI2cHw(I2C_BUS);
	u8g2.setI2CAddress(I2C_ADDRESS);
	u8g2.begin();

  u8g2.clearBuffer();

	u8g2.setFont(u8g2_font_6x13_tr);            // choose a suitable font
  u8g2.drawButtonUTF8(42, 50, U8G2_BTN_INV | U8G2_BTN_SHADOW2 | U8G2_BTN_HCENTER | U8G2_BTN_BW1, 0, 2, 2, "Yes");

  u8g2.sendBuffer();

	u8g2.doneI2c();
	u8g2.doneUserData();
  return 0;
}
```

```bash
make -j $(nproc)
arm-linux-gnueabihf-g++ -c -o obj/main.cpp.o main.cpp -I ./lib/u8g2/sys/arm-linux/drivers -I ./lib/u8g2/sys/arm-linux/drivers -I ./lib/u8g2/sys/arm-linux/port -I ./lib/u8g2/csrc -I ./lib/pixel-view/src -W -Wall -D __ARM_LINUX__ -L ./lib -lm
main.cpp: In function ‘int main()’:
main.cpp:16:8: error: ‘class U8G2_SH1106_128X64_NONAME_F_HW_I2C’ has no member named ‘drawButtonUTF8’; did you mean ‘drawExtUTF8’?
   16 |   u8g2.drawButtonUTF8(42, 50, U8G2_BTN_INV | U8G2_BTN_SHADOW2 | U8G2_BTN_HCENTER | U8G2_BTN_BW1, 0, 2, 2, "Yes");
      |        ^~~~~~~~~~~~~~
      |        drawExtUTF8
make: *** [Makefile:56: obj/main.cpp.o] Error 1
.
```

This is my project structure if it helps:
```
.
├── lib
│   └── u8g2 (cloned from https://github.com/olikraus/u8g2.git)
├── main.cpp
├── Makefile
```

ironlungx, 2025-03-02:
Yup, I've fixed it temporarily by copying the following code block into `port/U8g2lib.h`
```cpp
    /* u8g2_button.c */
    void drawButtonUTF8(u8g2_uint_t x, u8g2_uint_t y, u8g2_uint_t flags, u8g2_uint_t width, u8g2_uint_t padding_h, u8g2_uint_t padding_v, const char *text) { 
      u8g2_DrawButtonUTF8(&u8g2, x, y, flags, width, padding_h, padding_v, text); }
```

olikraus, 2025-03-03:
hmm ok, you fixed it... but usually the code should be there...

ironlungx, 2025-03-03:
I couldn't find it in the port file :(. unless I'm doing something wrong

olikraus, 2025-03-03:
Which port file?

ironlungx, 2025-03-07:
[this](https://github.com/olikraus/u8g2/blob/master/sys/arm-linux/port/U8g2lib.h) file. There's no `drawUTF8Button` method. sorry for late response

olikraus, 2025-03-09:
oh, I see. Those ports are just contributed codes from other people. I actually don't update them, thats why it is missing.
But I think taking over the code from the csrc dir should be the best.
Maybe I should add a readme into those directories.

ironlungx, 2025-03-09:
ah okay, that makes sense, if possible I'll make a PR 
thank you 🙏