olikraus/u8g2 #2604: u8x8 vs SMBUS

bbd666, 2025-03-05:
I'd like to translate the ST7588 driver to python.
Thanks to the datasheet the protocol is clear. 
One thing remains confusing to me.

In U8g2, to send I2C msgs to the LCD, 4 types of commands are used : 
  U8X8_START_TRANSFER(),             
  U8X8_C(),					
  U8X8_END_TRANSFER(),             	
  U8X8_END()       

I understand U8X8_C() is similar to write_byte_data. What about the 3 other commands?

olikraus, 2025-03-05:
> U8X8_C() is similar to write_byte_data.

Partly yes. Additionally U8X8_C() will ensure, that the data byte is interpreted as command for the display controller

> U8X8_START_TRANSFER()

Will ensure that the display controller is ready to receive commands. For example in SPI mode many display controllers expect a lowlevel on CS line, so U8X8_START_TRANSFER() will pull CS to low (but actzally this depends on the protocol)

> U8X8_END_TRANSFER(),

This will undo the U8X8_START_TRANSFER()

> U8X8_END()

This just marks the end of a command sequence.

bbd666, 2025-03-06:
ok 
Thks so much for you answer.
I believe the SMBUS command write_byte_data already embeds these start and stop triggers.