olikraus/u8g2 #2606: ST7567 (on JHDLCM) not supported

ceasarseptember, 2025-03-06:
Hello,

I have tried every constructor for the ST7567 in the graphics library and cannot get anyone of them to work properly on my display. Some cannot have the contrast dimmed and so nothing can be viewed on the display even by reducing or increasing the contrast value.

Using normal setting U8G2_R0 the display is upside down and words are reversed. Using U8G2__MIRROR, some of the constructors seem to work but the first 3 or 4 columns of the display are not displayed correctly and look like random data in these columns.

Is there any way around this or do I have to make my own initialize routine ?

Thanks.
cs

olikraus, 2025-03-06:
You can always use the sendF command to modify the display setup.
Is there any datasheet of your display? 
If you would ignore the contrast, which constructor produces correct output without using U8G2_MIRROR?
Any picture?

ceasarseptember, 2025-03-10:
Hello, thanks, I did eventually get the display working using the U8G2_ST7567_ENH_DG128064_F_4W_SW_SPI constructor BUT the display starts on pixel no 4 and I can show this by drawing a box frame outline using u8g2.drawFrame(0,0,u8g2.getDisplayWidth()-4,u8g2.getDisplayHeight() ); and unless I subtract 4 pixels from the width I will not see the right side of the box. Not sure if this is the display H/W.
Great work.
cs

ceasarseptember, 2025-03-10:
Hello, I could not find a reason why the 0(zero) for the start of the LCD display would be offset by +4 pixels. Is there anyway one can offset the start to be -4 pixels.

When drawing a frame or box : u8g2.drawFrame(0,0,u8g2.getDisplayWidth()-4,u8g2.getDisplayHeight() ); and unless I subtract 4 pixels from the width I will not see the right side of the box.
thanks
cs

olikraus, 2025-03-14:
You could adjust the offset here: https://github.com/olikraus/u8g2/blob/2b75f932b5ef4b4de8edf73e1a690702a35b1976/csrc/u8x8_d_st7567.c#L559-L560