olikraus/u8g2 #2610: RPi Pico W FLASH overflow

dwn-dragon, 2025-03-14:
u8g2 doesn't seem to work on the Pico 1 W even tho u8x8 works without issues.
I've successfully implemented and tested the needed HAL functions with u8x8 and the code both compiles and shows output on display.
If I then switch to u8g2, it won't show any kind of output using a u8x8 font.
If I try to use a u8g2 font, I get this error:
```
[build] [...]/.pico-sdk/toolchain/14_2_Rel1/bin/../lib/gcc/arm-none-eabi/14.2.1/../../../../arm-none-eabi/bin/ld.exe: Lathe.elf section `.rodata' will not fit in region `FLASH'
[build] [...]/.pico-sdk/toolchain/14_2_Rel1/bin/../lib/gcc/arm-none-eabi/14.2.1/../../../../arm-none-eabi/bin/ld.exe: region `FLASH' overflowed by 11482616 bytes
```
I've even tested it by trying to draw a circle, but no output on the display then either.

olikraus, 2025-03-14:
You need to enable the object garbage collector during linking stage. 
This usually is standarf Arduino IDE, but might be missed for boards without full Arduino support.

dwn-dragon, 2025-03-15:
The issue has been fixed by adding the following line (as stated in #2159): 
`set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -ffunction-sections -fdata-sections -Wl,--gc-sections")`
It should not be needed if u8g2 is defined as an interface library instead of a static library, it might be worth changing it for ease of use.

olikraus, 2025-03-15:
I am not responsible for any CMAKE related files in this repo. I just accept PR for them, but they are not supported by me. If you feel that they should be improved, please create a PR.

dwn-dragon, 2025-03-16:
I think it'd be useful to add a note to https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

closed 2025-03-16 by dwn-dragon