olikraus/u8g2 #2611: U8G2 SSD1309 not working

dwn-dragon, 2025-03-15:
While u8x8 does show output, nothing is drawn on the display if u8g2 is used instead.
I'm using an Arduino Nano 33 Ble Sense with PlatformIO and Arduino environment, this is the main:
```
#include <Arduino.h>

#include <Wire.h>
#include <U8g2lib.h>

U8G2_SSD1309_128X64_NONAME0_F_HW_I2C display { U8G2_R0, U8X8_PIN_NONE, PIN_WIRE_SCL, PIN_WIRE_SDA };

void setup() {
	display.begin();
	display.setFont(u8g2_font_ncenB14_tr);
	display.clear();
	display.home();
}

void loop() {
	display.clearBuffer();
	display.drawStr(0, 0, "Hello world!");
	display.sendBuffer();

	delay(1000);

	display.clearBuffer();
	display.sendBuffer();

	delay(1000);
}
```

olikraus, 2025-03-15:
Maybe one of the most common mistakes for u8g2. This has been discussed very very often :-D

Please have a look at the example picture: https://github.com/olikraus/u8g2/wiki/u8g2reference#drawstr

Ok, maybe I should add a third note, that (by default) the provided x/y position is the lower left edge of the chars.

**ToDo for me: Update Doc for drawStr**

closed 2025-03-15 by dwn-dragon

olikraus, 2025-03-15:
Wiki update done.