olikraus/u8g2 #2612: how to add an LCD with 320x160 RA8835

devicemodder, 2025-03-16:
Got an LCD  with a resolution of 320x160. how can i get that supported in the library?

olikraus, 2025-03-16:
There should be an existing constructor: U8G2_RA8835_320X240

Somehow the page https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#ra8835-320x240 is broken. I guess I need to check this.

devicemodder, 2025-03-16:
> There should be an existing constructor: U8G2_RA8835_320X240
> 
> Somehow the page https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#ra8835-320x240 is broken. I guess I need to check this.

Was looking for that earlier. Saw that there's no page there. Is there a way that I can modify the existing library to work with my display, or specify the resolution? 

Cause my display is actually 320x160.

olikraus, 2025-03-16:
oh, ok, issue is already there: https://github.com/olikraus/u8g2/issues/2557

olikraus, 2025-03-16:
> Cause my display is actually 320x160.

There is no simple way to specify the resolution, but I can create the constructor based on an existing one.
Will the U8G2_RA8835_320X240 show something on the screen? Any picture?

devicemodder, 2025-03-16:
Haven't had a chance to wire it up yet. I'll hook it up tomorrow, and test it out, then get back to you if that's OK.

devicemodder, 2025-03-17:
> > Cause my display is actually 320x160.
> 
> There is no simple way to specify the resolution, but I can create the constructor based on an existing one. Will the U8G2_RA8835_320X240 show something on the screen? Any picture?

Yes, i actually do get a picture 

[image]

olikraus, 2025-03-25:
ToDo: Add 320x160 constructor for RA8835