olikraus/u8g2 #2614: ESP32 running ST7920 128x64 on Hardware SPI VSPI

deeja, 2025-03-17:
So this was going to be a question, but now that it's working I thought I would put this here for posterity. 
I had quite a few issues. One being a blinking display caused using `CND` rather than `GND`. Why is it placed right by 5v on a NodeMCU 32?!
Another was confusing MISO and MOSI. 
Getting the Hardware SPI was also being annoying until I ran `SPI.begin(...)`.

So here is some code that runs the display over hardware SPI on the VSPI pins.
Enjoy.

 - GPIO18 
   - R/W 
   - SID
   - MOSI
 - GPIO 5
   - RS/CS 
   - CS
 - GPIO 23
   - SCLK
   - E

RST is not connected.
Code is taken from an ESP8266 tutorial that I got running quicker. https://forum.arduino.cc/t/how-to-connect-lcd-12864-st7920-to-nodemcu-esp8266-module/534077/2

```cpp
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R0, /* CS=*/ 5); // ESP32

void setup(void) {
  SPI.begin(18, -1, 23, 5); // Need to start the SPI 
  u8g2.begin();  
  u8g2.setBusClock(500000); // Artifacting can happen over this frequency
}

uint8_t m = 24;

void loop(void) {
  char m_str[3];
  strcpy(m_str, u8x8_u8toa(m, 2));		/* convert m to a string with two digits */
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_logisoso62_tn);
    u8g2.drawStr(50,63,m_str);
  } while ( u8g2.nextPage() );
  delay(1000);
  m++;
  if ( m == 60 )
    m = 0;
}
```

deeja, 2025-03-17:
Closing as this is solved

closed 2025-03-17 by deeja

olikraus, 2025-03-25:
Thanks a lot!