olikraus/u8g2 #2620: How transfer pin settings to setup()?

mister-Monk, 2025-03-25:
There is an ESP8266 and a SSD1309 display.
All GPIO settings are stored in a separate file.
The Wire library is launched after receiving GPIO numbers from SPIFFS, and services some sensors.
How can I change the sketch so that after reading SPIFFS I can pass GPIO numbers to the constructor?

`/*
  GraphicsTest.ino
  Universal 8bit Graphics Library (https://github.com/olikraus/u8g2/)
*/

//#include <Arduino.h>
#include <U8g2lib.h>

#include <Wire.h>

#define SCL_pin       0            //---------- I2C SCL 0
#define SDA_pin       2            //---------- I2C SDA 2

U8G2_SSD1309_128X64_NONAME0_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL_pin, /* data=*/ SDA_pin, /* reset=*/ U8X8_PIN_NONE);

  uint8_t ssd1309_i       = -1;     // ===== кубики
  bool    ssd1309_c       = true;   // ===== кубики color?

void setup(void)
{
  Wire.begin(SDA_pin, SCL_pin);   // For certain peripherals.
  u8g2.begin();
}

void loop(void)
{
  Draw_Box();
  u8g2.sendBuffer();
  delay(500);
}

void Draw_Box()
{
    if (++ssd1309_i > 7)
    {
      ssd1309_i = 0;
      ssd1309_c = !ssd1309_c;                 // Меняем цвет кубиков
    }
    if (ssd1309_c)  u8g2.setDrawColor(1);     // WHITE
    else            u8g2.setDrawColor(0);     // BLACK
    u8g2.drawBox(ssd1309_i * 16, 0, 15, 16);  // Рисуем кубики
}`

v1993, 2025-03-25:
Few ideas come to mind:

1. Make your display variable (`u8g2`) a pointer, then use a `u8g2 = new U8G2_SSD1309_128X64_NONAME0_F_SW_I2C(...)` in `Setup`. Requires dynamic memory allocation, so YMMV.
2. Come up with a way to use placement new if you want to avoid the dynamic allocation while also dodging automatic initialization. Probably use some kind of a `union`?
3. Use the generic `U8G2` class for `u8g2`, then call the [corresponding setup functions](https://github.com/olikraus/u8g2/blob/2b75f932b5ef4b4de8edf73e1a690702a35b1976/cppsrc/U8g2lib.h#L7412-L7417) manually. Use `u8g2.getU8g2()` and `u8g2.getU8x8()` to get corresponding internal C structs needed for these calls.

Out of these 3, the third option seems like the most reliable to me, although it's not quite as pretty as using a pre-made constructor.

mister-Monk, 2025-03-25:
Victory!
My carelessness is to blame for everything.
I tested the first variant yesterday.

<details>
  <summary>My mistake is here.</summary>

```c++

// Somewhere before everything...()...
U8G2_SSD1309_128X64_NONAME0_F_SW_I2C* u8g2;
.
.
.
void setup(void){}
.
.
.
void init_SSD1309(uint8_t SDA_pin, uint8_t SCL_pin)
{
// This is an incorrect line.
//u8g2 = new U8G2_SSD1309_128X64_NONAME0_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL_pin, /* data=*/ SDA_pin, /* reset=*/ U8X8_PIN_NONE);
// This line is correct.
  u8g2 = new U8G2_SSD1309_128X64_NONAME0_F_SW_I2C(U8G2_R0, /* clock=*/ SCL_pin, /* data=*/ SDA_pin, /* reset=*/ U8X8_PIN_NONE);

  u8g2->begin();
}

```

</details>