olikraus/u8g2 #2628: Ghost effect on GLCD

dariods8474, 2025-04-14:
I am using ST7920 Monochrome GLCD on ESP32 using U8G2 library. I get a ghost effect only on bold numbers, lines and box but not on normal fonts (u8g2_font_6x12_tf). The ghost effect reduces with reducing the contrast by rotating the trimpot on the back of the GLCD but reducing contrast is loss of clarity in daylight. Why only on thick prints? Power supply is battery.

olikraus, 2025-04-15:
Not sure how software can help here. Maybe the voltage boost circuit is broken.

olikraus, 2025-04-15:
You can adjust the x-offset here:
https://github.com/olikraus/u8g2/blob/82efd57daea8593d36194940f53fb7e518390276/csrc/u8x8_d_ssd1306_64x48.c#L190

For the y offset try this:
u8g2.sendF("c", 0x040 + y);
where y is the offset value.