olikraus/u8g2 #2631: SSD1363 8080 troubleshooting

Wheatley06, 2025-04-16:
Hello,
I'm having an issue with this display ([aliexpess link](https://www.aliexpress.com/item/1005006337009149.html?spm=a2g0o.order_list.order_list_main.73.1db81802OhbLtU), [screen documentation](https://midasdisplays.com/wp-content/uploads/2024/12/MDOG256128AV-YM.pdf)). It is 256x128 SSD1363 display, I connect it via 8080 parallel interface.

I have a working code ([sketch_apr13a_safe.zip](https://github.com/user-attachments/files/19767202/sketch_apr13a_safe.zip)) with screen initialization written manually, but I want it to work with u8g2. I wrote this code drawing vertical stripes:
`#include <U8g2lib.h>
#include <Arduino.h>

U8G2_SSD1363_256X128_1_8080 u8g2(U8G2_R0, 
  13, 14, 21, 47, 48, 45, 35, 36, // D0-D7
  11,  // WR
  9,   // CS
  10,  // DC
  46   // RESET
);

void setup() {
  u8g2.begin();
}

void loop() {
  u8g2.clearBuffer();

  for (int x = 0; x < 256; x += 4) {
    u8g2.drawBox(x, 0, 2, 128);
  }

  u8g2.sendBuffer();

  delay(1000);
}`

this is how it looks like:
[image]

To compare this is how it looks like with my code:
[image]

olikraus, 2025-04-16:
Maybe digitalWrite(12, HIGH); is missing...

Wheatley5925, 2025-04-16:
Thanks for the reply

I changed the setup function:

```
void setup() {
  pinMode(9, OUTPUT);
  pinMode(46, OUTPUT);
  pinMode(10, OUTPUT);
  pinMode(11, OUTPUT);
  pinMode(12, OUTPUT);

  digitalWrite(12, HIGH);
  u8g2.begin();
}
```

And after changing the class to `U8G2_SSD1363_256X128_F_8080` it works perfectly fine! Thanks

closed 2025-04-16 by Wheatley5925