olikraus/u8g2 #2636: SSD1306 128x64 I2C display offset one time

Joemanix65, 2025-04-27:
Hello, I'm learning how to display the value of some cryptocurrencies. The first display is always offset. Then everything is fine. Do you know why? Thank you.

[image]

[image]

olikraus, 2025-05-01:
To give a proper answer, we would need more details on the displays: Datasheet, resolution, display controller, u8g2 constructor...

Such offsets usually appear if the display is smaller than the memory in the display controller (for example a 64x48 OLED, which is driven by a SSD1306 controller with 128x64 memory)

Joemanix65, 2025-05-04:
hello, i have found the datasheet here is:

[GME12864-11-1-1.pdf](https://github.com/user-attachments/files/20026853/GME12864-11-1-1.pdf)

olikraus, 2025-05-07:
The constructor is correct for the display, assuming that the datasheet is matching.
Maybe there is an issue in your code. 
You could post your code here if you want.

Joemanix65, 2025-05-07:
thank you very much for your help and work, here is my code :
#include <U8g2lib.h>
#include <WiFi.h>
#include <HTTPClient.h>
#include <ArduinoJson.h>
#include "secrets.h"

U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

const char* ssid = SSID;
const char* password = WIFI_PASSWORD;
const char* coin1 = "ethereum";
const char* coin2 = "bitcoin";
const char* coin3 = "binancecoin";
const char* coin4 = "storj";

String url = (String)"https://api.coingecko.com/api/v3/simple/price?ids=" + coin1 + "," + coin2 + "," + coin3 + "," + coin4 + "&vs_currencies=eur&precision=2";
//const char* euro = "\xE2\x82\xAC";

void setup() {
  // put your setup code here, to run once:
  u8g2.begin();
  //u8g2.setContrast(85);
  delay(1000);

  Serial.begin(115200);

  WiFi.begin(ssid, password);

  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_9x15B_tf);
  u8g2.drawStr( 0, 28, "Connecting...");
  u8g2.sendBuffer();

  while(WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
}

void displayError(const char* error) {
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_9x15_tf);
  u8g2.drawStr( 0, 10, error);
  u8g2.sendBuffer();
}

void loop() {

 if(WiFi.status() == WL_CONNECTED) {
    HTTPClient http;
    http.begin(url);

    int httpResponseCode = http.GET();

    if(httpResponseCode > 0) {
      String payload = http.getString();
      StaticJsonDocument<200> doc;
      DeserializationError error = deserializeJson(doc, payload);

      if(!error) {

          float coin1Price = doc[coin1]["eur"];

          //displayPrice(coin1Price);
          u8g2.clearBuffer();
          u8g2.setFont(u8g2_font_10x20_tf);
          u8g2.setCursor(0, 2);
          u8g2.print(coin1);
          u8g2.setCursor(0, 32);
          u8g2.print(coin1Price);
          u8g2.setFont(u8g2_font_unifont_t_symbols);
          u8g2.setFontPosTop();
          u8g2.drawUTF8X2(75, 28, "€");
          u8g2.sendBuffer();
          delay(6000);

        float coin2Price = doc[coin2]["eur"];

          //displayPrice(coin2Price);
          u8g2.clearBuffer();
          u8g2.setFont(u8g2_font_10x20_tf);
          u8g2.drawStr(0, 2, coin2);
          u8g2.setCursor(0, 32);
          u8g2.print(coin2Price);
          u8g2.setFont(u8g2_font_unifont_t_symbols);
          u8g2.setFontPosTop();
          u8g2.drawUTF8X2(85, 28, "€");
          u8g2.sendBuffer();
          delay(6000);

        float coin3Price = doc[coin3]["eur"];

          //displayPrice(coin3Price);
          u8g2.clearBuffer();
          u8g2.setFont(u8g2_font_10x20_tf);
          u8g2.setCursor(0, 2);
          u8g2.print(coin3);
          u8g2.setCursor(0, 32);
          u8g2.print(coin3Price);
          u8g2.setFont(u8g2_font_unifont_t_symbols);
          u8g2.setFontPosTop();
          u8g2.drawUTF8X2(75, 28, "€");
          u8g2.sendBuffer();
          delay(6000);

        float coin4Price = doc[coin4]["eur"];

          //displayPrice(coin1Price);
          u8g2.clearBuffer();
          u8g2.setFont(u8g2_font_10x20_tr);
          u8g2.setCursor(0, 0);
          u8g2.print(coin4);
          u8g2.setCursor(0, 32);
          u8g2.print(coin4Price);
          u8g2.setFont(u8g2_font_unifont_t_symbols);
          u8g2.setFontPosTop();
          u8g2.drawUTF8X2(75, 28, "€");
          u8g2.sendBuffer();
          delay(6000);

      } else {
        displayError("JSON Error");
      }
      } else {
        displayError("HTTP Error");
      }

      http.end();
  } else {
    displayError("WiFi Error");
  }
  //delay(1000);
}

olikraus, 2025-05-18:
I saw some notifications, but nothing else here on this github issue.