olikraus/u8g2 #2640: Problem using HW in constructor with Teensy for several Display

Gelstronic, 2025-05-04:
Hi,
first of all, many thanks for this amazing library.

I use 10 several/difference displays for my midi-controller project, with my PiPico rp2040 all works fine with  HW use in the constructors:

```
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI display1(U8G2_R2, /*CS*/ 6,/*DC*/ 4, /* reset=*/5);

U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI display2(U8G2_R2, /* cs=*/ 7, /* dc=*/ 4,/*RST*/ U8X8_PIN_NONE);	
U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI display3(U8G2_R2, /*CS*/ 8,/*DC*/ 4, /*RST*/ U8X8_PIN_NONE);
U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI display4(U8G2_R2, /*CS*/ 9,/*DC*/ 4, /*RST*/ U8X8_PIN_NONE);
U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI display5(U8G2_R2, /*CS*/ 10,/*DC*/ 4, /*RST*/ U8X8_PIN_NONE);

U8G2_SSD1312_128X32_F_4W_HW_SPI display6(U8G2_R2, /*CS*/ 11,/*DC*/ 4, /*RST*/ U8X8_PIN_NONE);
U8G2_SSD1312_128X32_F_4W_HW_SPI display7(U8G2_R2, /*CS*/ 12,/*DC*/ 4, /*RST*/ U8X8_PIN_NONE);
U8G2_SSD1312_128X32_F_4W_HW_SPI display8(U8G2_R2, /*CS*/ 13,/*DC*/ 4, /*RST*/ U8X8_PIN_NONE);
U8G2_SSD1312_128X32_F_4W_HW_SPI display9(U8G2_R2, /*CS*/ 14,/*DC*/ 4, /*RST*/ U8X8_PIN_NONE);
U8G2_SSD1312_128X32_F_4W_HW_SPI display10(U8G2_R2, /*CS*/ 15,/*DC*/ 4, /*RST*/ U8X8_PIN_NONE);
```
Now i change to a Teensy 4.1 and nothing work, all Displays are  black.

What i try so far:

- lower the cpu frequency from 600Mhz to 150Mhz
- lower the BusClock to 1Mhz
-  change SPI mode from 0 to 3

No success, BUT when using  SW in the constructor all works fine and of course slooow...

I suspect that it is a timing problem, at some point the teensy is too fast, but where ?

Any thoughts ?

olikraus, 2025-05-07:
> Any thoughts ?

You probably need to figure out, which pins are used by the HW subsystem. Maybe the Teensy outputs the data on different gpio pins then expected?

Gelstronic, 2025-05-08:
I've checked that several times, the default HW GPIOs for SPI are 13 for Clock and 11 for MOSI (Teensy4.1)
https://www.pjrc.com/teensy/pinout.html

Gelstronic, 2025-05-11:
Strange things happen, I have lowered the teensy cpu clock even further to 24 Mhz and now 8 out of 10 displays are working