olikraus/u8g2 #2642: Problem in Creating new font for u8g2

Arashfzn, 2025-05-10:
Hi Oliver; 
I hope you are well
I need to have a 20 pixel and 24 pixel Persian font for u8g2 , I have a Persian_Reshaper that works well with u8g2 persian fonts like : u8g2_font_samin_16_t_all
I downloaded the unicode zar normal font, "zar.ttf" font from : azfonts.net 
https://www.azfonts.net/fonts/zar/normal-262814

[zar.zip](https://github.com/user-attachments/files/20137829/zar.zip)

I also realized that in order for this font to be compatible with the Persian Reshaper fonts in the library (like "samim"), the glyph maps must be the same as those fonts. so I make a persian_map.txt file :

[persian_map.txt](https://github.com/user-attachments/files/20137862/persian_map.txt)

map 193=1569
map 194=1570
map 195=1571
map 196=1572
map 197=1573
map 198=1574
map 199=1575
map 200=1576
map 555=1577
map 202=1578
map 203=1579
map 204=1580
map 205=1581
map 206=1582
map 207=1583
map 208=1584
map 209=1585
map 210=1586
map 184=1688
map 211=1587
map 212=1588
map 213=1589
map 214=1590
map 215=1591
map 216=1592
...

then I tryed this commands:

otf2bdf -r 72 -p zar.ttf -o zar20.bdf
bdfconv -v -f 1 -m persian_map.txt -n u8g2_font_zar20 -o zar20.c zar20.bdf

but unfortunately I dont get any gliph in output .c file:

/*
  Fontname: -FreeType-Zar-Medium-R-Normal--20-200-72-72-P-85-ISO10646-1
  Copyright: Generated by Sinasoft Co.
  Glyphs: 0/460
  BBX Build Mode: 0
*/
const uint8_t u8g2_font_zar20[27] U8G2_FONT_SECTION("u8g2_font_zar20") = 
  "\0\0\2\2\0\0\0\0\0\0\0\0\0\17\375\15\377\0\0\0\0\0\2\0\0\0";

Would you please help me, I tryed so many times with so many ways.
thank you very much.

olikraus, 2025-05-18:
The syntax of the mapfile is wrong. You need to follow the instructions given in the commandline help from bdfconv.exe:
```
map := <mapcmd> { "," <mapcmd> }
mapcmd := <default> | <maprange> | <exclude> | <exclude-kerning>
default := "*"
maprange := <range> [  ">" <addexpr> ]        Move specified glyph <range> to target code <num>
exclude := "~" <range> 
exclude-kerning:= "x" <range> 
range := <addexpr> [ "-" <addexpr> ]          Select glyphs within specified range
addexpr := <mulexpr> [ "+" <mulexpr> ]
mulexpr := <num> [ "*" <num> ]
num := <hexnum> | <decnum>
hexnum := "$" <hexdigit> { <hexdigit> }
decnum := <decdigit> { <decdigit> }
decdigit := "0" | "1" | "2" | "3" | "4" | "5" | "6" | "7" | "8" | "9"
hexdigit := "a" | "b" | "c" | "d" | "e" | "f" | "A" | "B" | "C" | "D" | "E" | "F" | <decdigit>
{ } zero, one ore more, [ ] zero or once, | alternative
```

Note: The map file not only provides the mapping but also describes, which chars should be used. So please ensure, that all required chars are mentioned in the map file.

Let look into your file.
It starts like this:
```
map 193=1569
map 194=1570
map 195=1571
map 196=1572
...
```  
I am not exactly sure what this means, but 193 is "Á" and 1569 is "ᕩ" (Canadian Syllabics West-Cree Thwa). I am not sure whether this mapping makes sense, but lets assume you want to map glyph code 193 to glyph code 1569, then the correct syntax is
```
193>1569
...
If you need to do this for multiple codewords, you need to add "," in between:
```
193>1569,
194>1570,
195>1571,
196>1572,
...
Additionally you can change the mapping for a complete range, so for the above four lines you could do this:
```
193-196>1569,
...
```

Hope this helps.

Arashfzn, 2025-05-19:
Hi Oliver;
I'm really sorry to bother you, I know you're very busy.

[image]
the image above shows the maping of my zar20.bdf
I also changed the persian_map.txt to this format:
193>1569,194>1570,195>1571,196>1572,197>1573,198>1574, . . . 
But unfortunately I don't get any Gligh 
zar20.c  output:

/*
  Fontname: -FreeType-Zar-Medium-R-Normal--20-200-72-72-P-85-ISO10646-1
  Copyright: Generated by Sinasoft Co.
  Glyphs: 0/460
  BBX Build Mode: 0
*/
const uint8_t u8g2_font_zar20[27] U8G2_FONT_SECTION("u8g2_font_zar20") = 
  "\0\0\2\2\0\0\0\0\0\0\0\0\0\17\375\15\377\0\0\0\0\0\2\0\0\0";

bdfconv report:

D:\Myprojects\otf2bdf>bdfconv -v -f 1 -m persian_map.txt -n u8g2_font_zar20 -o z
ar20.c zar20.bdf
Parse File zar20.bdf: 460 glyph(s) found
Map: map_cmd_list='persian_map.txt'
Map: exclude=0 from=0/$0 to=0/$0 map=0/$0
Reduce: Start
Reduce: End
CalculateMaxBBX: x=0, y=0, w=0, h=0
CalculateMaxBBX: Encodings x=0, y=0, w=0, h=0
bf_CalculateMinMaxDWidth: dx_min=32767, dx_max=-32767
bf_CalculateMinMaxDWidth: x_min=32767, x_max=-32767
bf_CalculateMaxBitFieldSize: bbx.x=0, bbx.y=0, bbx.w=0, bbx.h=0, dwidth=0
RLE Compress: best zero bits 2, one bits 2, total bit size 0
RLE Compress: Font code generation, selected glyphs=0, total glyphs=460
RLE Compress: 'A' pos = 0, 'a' pos = 0
RLE Compress: Font size 27
bf_WriteU8G2CByFilename: Write file 'zar20.c'

olikraus, 2025-06-10:
according to your screenshot the original char is at 1576, so the mapping (if required) needs to be in the opposite direction: 1569>193 and so on.

closed 2025-06-23 by Arashfzn