olikraus/u8g2 #2643: SH1122 not rendering correctly sometimes

balinmitchell, 2025-05-11:
I am having an issue with an SH1122 256x64 OLED where sometimes the image on the screen is not what has been sent. This fault is not consistent as sometimes it works perfectly. I am updating the display at 100 Hz and have tried page mode as well as full frame buffer. I am using a Teensy4.1 so computing power shouldn't be an issue.

[image]

balinmitchell, 2025-05-12:
Some more information to help troubleshoot this one. This issue only occurs sometimes and will go away if I keep cycling power until it displays correctly. It also gets worse when I add more data to the buffer. Have also tried pages but it doesn't help. 

`#include <Arduino.h>
#include <FlexCAN_T4.h>
#include <U8g2lib.h>
#include <SPI.h>

FlexCAN_T4<CAN1, RX_SIZE_256, TX_SIZE_16> can0;
CAN_message_t msg;

// U8G2_SSD1362_256X64_F_4W_HW_SPI u8g2(U8G2_R0, 10, 9, 8);
U8G2_SH1122_256X64_F_4W_HW_SPI u8g2(U8G2_R0, 10, 9, 8);

uint16_t tempCoolant, tempOil, tempGear, tempExterior, levelFuel, year;
uint8_t hours, minutes, seconds, day, month;
float batVoltage, speed, tps;
long timePrev;

#define NUM_RX_MAILBOXES 4

void canSniff(const CAN_message_t &msg);

void setup(void) {
  Serial.begin(9600); 
  delay(400);

  can0.begin();
  can0.setBaudRate(500000);

  u8g2.begin();
  u8g2.clearBuffer();

  
}

void loop() {

  can0.events();
  u8g2.firstPage();

  do {
    u8g2.setFont(u8g2_font_VCR_OSD_tu);
    u8g2.drawStr(0,16,"WTRTMP:");
    u8g2.setCursor(90,16);
    u8g2.print(tempCoolant);
    u8g2.drawStr(0,40,"GBXTMP:");
    u8g2.setCursor(90,40);
    u8g2.print(tempGear);
    u8g2.drawStr(0,63,"EXTTMP:");
    u8g2.setCursor(90,63);
    u8g2.print(int(tempExterior));
    // u8g2.drawStr(128,16,"SPEED:");
    // u8g2.setCursor(228,16);
    // u8g2.print(int(speed));
    // u8g2.drawStr(128,40,"TPS:");
    // u8g2.setCursor(228,40);
    // u8g2.print(int(tps));
    // u8g2.drawStr(128,63,"FUEL:");
    // u8g2.setCursor(228,63);
    // u8g2.print(int(levelFuel));
  } while (u8g2.nextPage()); 
  

  while((micros()-timePrev) < 10000); //100Hz Loop Rate
  timePrev = micros();
}`

olikraus, 2025-05-18:
Looks ok to me. However if the variables in the code are updated in parallel, then you need to double buffer them, because during the page mode, the variables must not change.
For example "tempExterior" should not change during the while loop. To solve this, introduce a new variable (like displayTempExterior) and assign a value to this variable at the beginning of the loop:
```
displayTempExterior = tempExterior;
```
Then inside the loop use:
```
u8g2.print(int(displayTempExterior));
```
Of course, by using the full buffer mode, this double buffering of the variables is not required.

balinmitchell, 2025-05-19:
I have tried with full buffer mode and have the same issue.
For the most part this issue happens during startup but occasionally will happen during running.

balinmitchell, 2025-10-05:
I am happy to have finally found the solution, by setting the bus speed with   u8g2.setBusClock(10000000); I have achieved some stability, I started with 1MHz but moved up to 10MHz with no issue.

closed 2025-10-05 by balinmitchell