olikraus/u8g2 #2644: Help with choosing class

jacekgajek, 2025-05-12:
I have this Arduino shield: https://wiki.dfrobot.com/LCD12864_Shield_SKU_DFR0287
attached to Arduino R4 WiFi board.

[image]

I can't figure out which class should I use. I tried several but I don't if it's error in constructor or somewhere else.

According to docs, it uses D7, D8, D9, D10, D11, D13, A0 pins from which D7 is for backlit and A0 for joystick.

I use VS Code with PlatformIO. I tried this (and several other constructors):
```
#include <Arduino.h>
#include <U8g2lib.h>

// I should use HW (hardware I guess) for better performance, right?
// U8G2_ST7565_NHD_C12864_1_3W_HW_SPI u8g2(U8G2_R0, PIN_D10, PIN_D8);

U8G2_ST7565_NHD_C12864_1_3W_SW_SPI u8g2(U8G2_R0, D13, D11, D10, D8);

void setup()
{
    u8g2.begin();
    u8g2.setContrast(0);

}

void loop()
{

    u8g2.clearBuffer();
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(0, 20, "Hello World!");
    u8g2.sendBuffer();
}
```
But it doesn't work (display is blank).

I can see that
```
U8G2_ST7565_NHD_C12864_1_4W_SW_SPI(rotation, clk, data, cs, dc [, reset])
U8G2_ST7565_NHD_C12864_2_4W_SW_SPI(rotation, clk, data, cs, dc [, reset])
U8G2_ST7565_NHD_C12864_F_4W_SW_SPI(rotation, clk, data, cs, dc [, reset])
```
have exact same signature, is there any difference among them?

v1993, 2025-05-12:
To answer your second question: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#buffer-size

TL;DR: RAM vs CPU tradeoff by limiting how many pixels are stored in memory.

jacekgajek, 2025-05-13:
Thanks. I tried both options.

I guess I will just use the old U8Glib, I just need to [add wiring_private.h manually](https://forum.arduino.cc/t/compilation-error-wiring-private-h-no-such-file-or-directory/1146752/4).

I'd really prefer to use the supported version of the library though.

jacekgajek, 2025-05-13:
For reference, this is code which works with U8glib.h:

```
U8GLIB_NHD_C12864 u8g2( D13, D11, D10, PIN_A0, PIN_D8);

void draw(void)
{
    u8g2.setFont(u8g_font_unifont);
    u8g2.drawCircle(64, 32, 20);
    u8g2.drawStr(0, 10, "Hello World!");
}

void setup()
{
    u8g2.begin();
    u8g2.setContrast(0); // Config the contrast to the best effect
}

void loop()
{
    u8g2.firstPage();
    do
    {
        draw();
    } while (u8g2.nextPage());
}
```

olikraus, 2025-05-18:
```
For U8G2_ST7565_NHD_C12864_1_4W_SW_SPI
 uint8_t clock -> SPI Interface: SCK
 uint8_t data -> SPI Interface: MOSI
 uint8_t cs -> SPI Interface: CS
 uint8_t dc -> SPI Interface: CD <--- not sure about this
 uint8_t reset -> SPI Interface: RST
```

DC or CD should be the same (it meens DATA/COMMAND or COMMAND/DATA), so the mapping is correct and the constructor should be:

```
 uint8_t clock -> SPI Interface: SCK -->13
 uint8_t data -> SPI Interface: MOSI-->11
 uint8_t cs -> SPI Interface: CS-->10
 uint8_t dc -> SPI Interface: CD <--- not sure about this -->9
 uint8_t reset -> SPI Interface: RST-->8
```

So the constructor should be:
```
U8G2_ST7565_NHD_C12864_1_4W_SW_SPI(U8G2_R0, D13, D11, D10, D9, D8);
```

BTW: in the example above, you have used the 3W SPI constructor: This will not work at all, because the display is (probably) configured in 4W SPI mode.

jacekgajek, 2025-05-19:
Not sure what example do you refer to, from my OP with u8g2 or the last message from Mar 13 with old library. The latter works perfectly well (it's running right now on my desk).

I've already tried U8G2_ST7565_NHD_C12864_1_4W_SW_SPI, it doesn't work. Screen is just blank:

[image]

```
#include <Arduino.h>
#include <U8g2lib.h>

U8G2_ST7565_NHD_C12864_1_4W_SW_SPI u8g2(U8G2_R0, D13, D11, D10, D9, D8);

void draw(void)
{
    u8g2.setFont(u8g_font_6x13r);
    auto time = "123456"; // myClock.getTime();
    u8g2.drawStr(0, 10, time);
}

void setup()
{
    u8g2.begin();
    u8g2.setContrast(0);
}

void loop()
{

    u8g2.firstPage();
    do
    {
        draw();
    } while (u8g2.nextPage());

    delay(100);
}
```