olikraus/u8g2 #2660: Can't get `setI2CAddress()` to work

christopher-ogibio, 2025-06-13:
```
#include <Arduino.h>
#include <U8g2lib.h>
#include <Wire.h>

U8G2_SSD1306_128X64_NONAME_F_SW_I2C display1(U8G2_R0, /* clock=*/SCL, /* data=*/SDA, /* reset=*/U8X8_PIN_NONE);
U8G2_SSD1306_128X64_NONAME_F_SW_I2C display2(U8G2_R0, /* clock=*/SCL, /* data=*/SDA, /* reset=*/U8X8_PIN_NONE);

void setup(void) {
  display2.setI2CAddress(0x7A);
  display2.begin();
  display2.setFont(u8g2_font_t0_14b_mf);  // choose a suitable font
  display2.clearDisplay();
  display1.setI2CAddress(0x78);
  display1.begin();
  display1.setFont(u8g2_font_t0_14b_mf);  // choose a suitable font
  display1.clearDisplay();
}

void loop(void) {
  display1.setDrawColor(1);
  display1.setCursor(2, 14);
  display1.print("test1");  // write something to the internal memory
  display1.sendBuffer();     // transfer internal memory to the display
  
  display2.setDrawColor(1);
  display2.setCursor(2, 40);
  display2.print("test2");  // write something to the internal memory
  display2.sendBuffer();      // transfer internal memory to the display
  delay(1000);
}
```

I have two SSD1306 128x64 displays both wired to the I2C bus on an Arduino Due. The displays work fine individually when not using `setI2CAddress()`. However, when I use the above code from https://forum.arduino.cc/t/u8glib-with-2-screen-and-i2c-address/681184/5 both displays print "test1" and "test2" in there respective cursor positions when I need them to print on each display separately. 
I have tried using the Adafruit I2C scanner code (https://learn.adafruit.com/scanning-i2c-addresses/arduino) but that shows no devices even when they both definitely work.
I have also tried using the I2C addresses: 0x3C, 0x3D, 0x3C * 2, and 0x3D * 2.

Any suggestions or help would be greatly appreciated!

christopher-ogibio, 2025-06-13:
I had bridged the solder pads on the back of one of the displays to change it's address. When I run the code with all of the `display2` lines commented out each display still shows the same thing. If I change the address of `display1` to `0x7A` then neither display updates.

christopher-ogibio, 2025-06-13:
I hadn't cut the track all the way to set the address. They now respond only to their individual addresses when only one display is called in the code. Using the above code both screen still display both lines of text.

olikraus, 2025-06-13:
> If i clear the buffer after each sendBuffer() that appears to have fixed the problem. Thank you for your help!

u8g2 tries to reduce RAM consumption: Both display objects will share the same RAM. 

> Oh nvm. Both displays have stopped working entirely now :(

Usually those SSD1306 displays are quite robust... maybe just a small wiring issue.

christopher-ogibio, 2025-06-16:
I've got them working now. Thank you again for your help @olikraus !