olikraus/u8g2 #2661: Feature Request: Add Support for ST7305 Controller (1.54", 200x200 Monochrome Display)

ischenz, 2025-06-13:
Hi @olikraus,

First, thank you for your excellent work on the u8g2 library — it's incredibly helpful for embedded display development!

I'd like to request support for the ST7305 display controller, used in a 1.54-inch, 200x200 resolution, monochrome reflective LCD module. This type of screen is ideal for ultra-low-power applications.

Display Details
Size: 1.54 inch

Resolution: 200×200 pixels

Color: Monochrome (full reflective)

Controller: ST7305

Interface: SPI

Power: Very low consumption — ideal for always-on or low-power applications

Datasheet: [YDP_154_H008_V3.pdf (Official module datasheet)](https://admin.osptek.com/uploads/YDP_154_H008_V3_c24b455ff9.pdf)

The display is used in various low-power wearable and IoT devices. Its controller (ST7305) appears to be somewhat similar to ST7565/ST75256, with some unique commands and memory addressing behavior.

Why this would be valuable
Adding support for the ST7305 would open up compatibility with a range of modern low-power displays. The 200x200 resolution is relatively uncommon in existing supported controllers, and this screen type performs particularly well in sunlight and ultra-low-power environments.

Willing to help
I have the hardware in hand and would be happy to help test, debug, or validate a prototype driver if needed. I’m also open to contributing code if given some guidance on u8g2’s internal driver structure.

Thanks again for maintaining such a great library!

Best regards,
chen

[image]

ischenz, 2025-06-13:
Sure, [](https://item.taobao.com/item.htm?id=857239043745&pisk=gOX7OiVWC4074KvKAQqqhyYBQc9BdoyapDtdjMHrvLpJOvIP5HkeT7xIdEsTqUP3Zw1fJZAyyeRedH_wogkUqgrQdMI9yp5yzHQVRZcPz98FuJsG5QkPv9oliNStUTPkLv9kKpUa7Ryw4gvHphPgqW3uHhxLLHppvLvx7Uiz6RyNqiiy2lSYQ9P0_3Ko9vQpwI3vfEK-w3d8cIKwvpHJ9YhxkEYvpHhJvjIvvHuKJHQKDxK9YYnJpeLxkhx6peQpJKEXxEKppphrQqtUNHI74PHgMxqnXg8jpvBXDyYOVVMHDtKSgFjWlPefhAAWWgLjpzpi-LTBoOU4Gib1wa-cRRaXKZfdJH_xkyRlvs_XVazs-eWl49TPV7UAcKYWHTCsvvIXFUAApI07lFBlDOvvgRE1DtbPqtsEvJKV7EIlepwTjL9JyKtcLrDJRsCdUQvnyPRlvs_XVdIy-A81QjGIcCDBcFrbcXcnG4312-bPY9dJm3eacogjtQKDcFrbcXcHwnxR_oZjlXf..&skuId=5929069279728&spm=tbpc.boughtlist.suborder_itemtitle.1.21bd2e8dcvTMwt)

ischenz, 2025-06-13:
Thank you for your reply. 
However, my screen is directly connected to the PCB (STM32 controller) and there is no SPI communication interface for connection, so it cannot be connected to the Arduino.

[image]

NoRolex, 2025-06-15:
Yes, I can confrim that the constructor works exactly the same as the ST7302.

Also i need to mention that i only use an FPC adapter and direct wires to the ESP32. No capacitors, resistors or any other recommended components. It seems to work fine like this.

olikraus, 2025-06-16:
@NoRolex Thanks for the feedback. I have updated the code to get the lines correct (beta 2.36.8). Will this work?
Can you send a picture of the same ino file? https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/ST7305Test/ST7305Test.ino

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

ischenz, 2025-06-19:
I made a PCB to connect my screen to the ESP32, and that got it working properly. This is how the code runs now:

[image]

ischenz, 2025-06-19:
I have the initialization code provided by the manufacturer. I hope it will be helpful to you.

`void Initial_ST7302(void)
{
        // 7305
	Write_Register(0xD6); //NVM Load Control 
	Write_Parameter(0X17); 
	Write_Parameter(0X02);
	 
	Write_Register(0xD1); //Booster Enable 
	Write_Parameter(0X01); 

	Write_Register(0xC0); //Gate Voltage Setting 
	Write_Parameter(0X08); //VGH 00:8V  04:10V  08:12V   0E:15V
	Write_Parameter(0X02); //VGL 00:-5V   04:-7V   0A:-10V 

        // VLC=3.6V (12/-5)(delta Vp=0.6V)		
	Write_Register(0xC1); //VSHP Setting (4.8V)	
	Write_Parameter(0X19); //VSHP1 	
	Write_Parameter(0X19); //VSHP2 	
	Write_Parameter(0X19); //VSHP3 	
	Write_Parameter(0X19); //VSHP4	
		
	Write_Register(0xC2); //VSLP Setting (0.98V)	
	Write_Parameter(0X31); //VSLP1 	
	Write_Parameter(0X31); //VSLP2 	
	Write_Parameter(0X31); //VSLP3 	
	Write_Parameter(0X31); //VSLP4 	
		
	Write_Register(0xC4); //VSHN Setting (-3.6V)	
	Write_Parameter(0X19); //VSHN1	
	Write_Parameter(0X19); //VSHN2 	
	Write_Parameter(0X19); //VSHN3 	
	Write_Parameter(0X19); //VSHN4 	
		
	Write_Register(0xC5); //VSLN Setting (0.22V)	
	Write_Parameter(0X27); //VSLN1 	
	Write_Parameter(0X27); //VSLN2 	
	Write_Parameter(0X27); //VSLN3 	
	Write_Parameter(0X27); //VSLN4

	Write_Register(0xD8); //HPM=32Hz
	Write_Parameter(0XA6); //~51Hz
	Write_Parameter(0XE9); //~1Hz

/*-- HPM=32hz ; LPM=> 0x15=8Hz 0x14=4Hz 0x13=2Hz 0x12=1Hz 0x11=0.5Hz 0x10=0.25Hz---*/
	Write_Register(0xB2); //Frame Rate Control 
	Write_Parameter(0X12); //HPM=32hz ; LPM=1hz 

	Write_Register(0xB3); //Update Period Gate EQ Control in HPM 
	Write_Parameter(0XE5); 
	Write_Parameter(0XF6); 
	Write_Parameter(0X05); //HPM EQ Control 
	Write_Parameter(0X46); 
	Write_Parameter(0X77); 
	Write_Parameter(0X77); 
	Write_Parameter(0X77); 
	Write_Parameter(0X77); 
	Write_Parameter(0X76); 
	Write_Parameter(0X45); 

	Write_Register(0xB4); //Update Period Gate EQ Control in LPM 
	Write_Parameter(0X05); //LPM EQ Control 
	Write_Parameter(0X46); 
	Write_Parameter(0X77); 
	Write_Parameter(0X77); 
	Write_Parameter(0X77); 
	Write_Parameter(0X77); 
	Write_Parameter(0X76); 
	Write_Parameter(0X45); 

	Write_Register(0x62); //Gate Timing Control
	Write_Parameter(0X32);
	Write_Parameter(0X03);
	Write_Parameter(0X1F);

	
	Write_Register(0xB7); //Source EQ Enable 
	Write_Parameter(0X13); 

	Write_Register(0xB0); //Gate Line Setting 
	Write_Parameter(0X32); //200 line

	Write_Register(0x11); //Sleep out 
	HAL_Delay(1000);

	Write_Register(0xC9); //Source Voltage Select  
	Write_Parameter(0X00); //VSHP1; VSLP1 ; VSHN1 ; VSLN1
	
	Write_Register(0x36); //Memory Data Access Control
	Write_Parameter(0X48); //MX=1 ; DO=1 	//48

	Write_Register(0x3A); //Data Format Select 
	Write_Parameter(0x10); //10:4write for 24bit ; 11: 3write for 24bit

	Write_Register(0xB9); //Gamma Mode Setting 
	Write_Parameter(0X20); //20: Mono 00:4GS 

	Write_Register(0xB8); //Panel Setting 
	Write_Parameter(0X29); // Panel Setting Frame inversion  09:column 29:dot_1-Frame 25:dot_1-Line

	//WRITE RAM 200*200
	Write_Register(0x2A); //Column Address Setting 
	Write_Parameter(0x16);//10 
	Write_Parameter(0x26); //38

	Write_Register(0x2B); //Row Address Setting 
	Write_Parameter(0X00); 
	Write_Parameter(0X63); //63
/*
	Write_Register(0x72); //de-stress off 
	Write_Parameter(0X13);
*/
	Write_Register(0x35); //TE
	Write_Parameter(0X00); //

	Write_Register(0xD0); //Auto power dowb
	Write_Parameter(0XFF); //

	Write_Register(0x39); //LPM

	Write_Register(0x29); //DISPLAY ON  
	
	  
}	

/*====================================================*/
void address(void)
{             
	Write_Register(0x2A); //Column Address Setting 
	Write_Parameter(0x16);//10
	Write_Parameter(0x26); //38
		
	Write_Register(0x2B); //Row Address Setting 
 	Write_Parameter(0X00); 
	Write_Parameter(0X63); //63
    
    Write_Register(0x2C);   //write image data
 }	
 
/*====================================================*/
  
void Write_Register(unsigned int idat)
{
	uint8_t data = idat;
	HAL_GPIO_WritePin(LCD_DC_GPIO_Port, LCD_DC_Pin, GPIO_PIN_RESET);
	HAL_SPI_Transmit(&hspi1, &data, 1, 100);
}

/*====================================================*/

void Write_Parameter(unsigned int ddat)
{
	uint8_t data = ddat;
	HAL_GPIO_WritePin(LCD_DC_GPIO_Port, LCD_DC_Pin, GPIO_PIN_SET);
	HAL_SPI_Transmit(&hspi1, &data, 1, 100);
}
`

olikraus, 2025-06-19:
The init code is not the problem. My main issue is the mapping between the data stream and the memory (bit) cells of the display.

It is mainly about this code:
https://github.com/olikraus/u8g2/blob/4abaf19f02662a71cbbdd935c8ecc2b5589d96f6/csrc/u8x8_d_st7305.c#L267-L303

If someone would be able to tell which values are need after the "=" sign so that the lines and the "Hello World" are correct, that would be too cool.

You can locate the file u8x8_d_st7305.c in your local copy of u8g2. Feel free to modify it.
I guess it will be some kind of trial and error process until all the assignments are correct.
Actually also the datasheet doesn't help much here.

ischenz, 2025-06-19:
Thank you very much for sharing this important information with me. I'm more than willing to give it a try. Could you please tell me what the values in this array represent?

olikraus, 2025-06-19:
Well even to me, it is not so clear, but let me try to explain this.

[image]

In u8g2 there is a linear sequence of bytes from left to right, where the most significant bit of each byte is on the left.
Now according to the datasheet, this data is transfered to the display RAM:

So if P1 is the uppermost bit in the first byte (for example the pixel of the upper left corner in u8g2), then the pixel will appear here also in the upper left corner of the display.

[image]

However, this is not true for the second pixel (right pixel of the upper left pixel in u8g2). This seems to be below the upper left pixel in the display.

In other words, the ST7503 seems to distribute the linear sequence of pixel in the display RAM into two rows of the display.

In the code we have to undo this pixel mixing of the display. So for P2 from above: If the second pixel in the display ram (bit 6 of the first byte) ends up in first pixel of the second row, then we have to invert this operation and place the first pixel of the second row from u8g2 into the bit position 6 of the first bit of the first line. Ahh.. not sure whether you get the point.

Maybe it is actually better to use this code:
https://github.com/olikraus/u8g2/blob/4abaf19f02662a71cbbdd935c8ecc2b5589d96f6/csrc/u8x8_d_st7305.c#L88

This will actually take each bit from the u8g2 input (p ponter) and assign it to the display memory of the display (buf).

All in all the problem is, that the display rearranges the bits and we have to undo this operation. But its beyond my head to got it done in theory. I have to sit and see what happens and how to shift the pixel from here to there so that the correct undo algorithm is created step by step.

olikraus, 2025-06-22:
Wow, this is truly amazing. Let me try to continue from here-

olikraus, 2025-06-22:
As a next step i will try to add the U8G2_ST7305_200X200_1_4W_HW_SPI constructor.

ischenz, 2025-06-22:
Thank you so much! I really appreciate your help.

ischenz, 2025-06-22:
Okay, I made the following changes and got this result. The display is no longer mirrored, but it seems to be rotated 180°, and the coordinates might still be off.
`U8X8_CA(0x36, 0xa4), 			// Memory Control 
u8x8_cad_SendArg(u8x8, 0x16);   // specific for the 122x250 LCD --> probably needs to be changed for the 
u8x8_cad_SendArg(u8x8, 0x26 );
`

[image]

olikraus, 2025-06-22:
Very good. I think now we need to adjust the view area. I have added a frame around the display so that it helps us to locate the visible area of the display:

https://github.com/olikraus/u8g2/blob/41971c801d520bcd2b29171566978bbf2592675c/sys/arduino/u8g2_page_buffer/ST7305Test/ST7305Test.ino#L435

I still feel that this line also needs some adjustment:
https://github.com/olikraus/u8g2/blob/41971c801d520bcd2b29171566978bbf2592675c/csrc/u8x8_d_st7305.c#L340

Can you send a picture with the frame from the updated example above?

ischenz, 2025-06-22:
Okay,I added the following code, and it’s now toggling between these two images repeatedly.
```
y+=0;
u8g2.drawFrame(0,0, 200,200);   
```

[image]
[image]

olikraus, 2025-06-22:
You mean with y+=115 it had been stable? Strange.. It could be also some other value between 1 and 114...

But there might be another issue, we probably need to change the B0 command:

https://github.com/olikraus/u8g2/blob/41971c801d520bcd2b29171566978bbf2592675c/csrc/u8x8_d_st7305.c#L203

According to your init sequence from above, it should be:
```
Write_Register(0xB0); //Gate Line Setting 
Write_Parameter(0X32); //200 line
```
So I assume for u8g2 we need 0x32 instead of 0x64

However it may happen that the constrast will change.

closed 2025-06-22 by olikraus

ischenz, 2025-06-22:
Okay,Here’s the code I changed.It’s still flickering between the two images.
```
U8X8_CA(0xB0, 0x32), 
y+=115;
``` 

[image]
[image]

olikraus, 2025-06-22:
Maybe in the column command, we not only have to change the first arg (to 0x16) but also second arg (which is 0x3a  at the moment).

https://github.com/olikraus/u8g2/blob/cffcd326825cbc2899391e894552ab91c3678a33/csrc/u8x8_d_st7305.c#L463-L465

at least in your init code from above we see this:

```
//WRITE RAM 200*200
Write_Register(0x2A); //Column Address Setting 
Write_Parameter(0x16);//10 
Write_Parameter(0x26); //38

Write_Register(0x2B); //Row Address Setting 
Write_Parameter(0X00); 
Write_Parameter(0X63); //63
```

So instead of 0x3a, it could be 0x26. It might be also 0x2f (or 0x2e or 0x30). My understand is this: The left offset is 0x16 (which is 0x16 == 22). The width is 200 pixel = 25 bytes, wo we have 22+25 = 47, which should be 0x2f for the second argument of the column address setting.

ischenz, 2025-06-22:
The code I'm currently running is as follows: 
```
        u8x8_cad_SendCmd(u8x8, 0x2a);   // column address set
        u8x8_cad_SendArg(u8x8, 0x16);   // specific for the 122x250 LCD --> probably needs to be changed for the 200x200
        u8x8_cad_SendArg(u8x8, 0x26 );
```
This part should be 0x2A, 0x16, 0x26, because 0x16 to 0x26 equals 17 (17 × 12 = 204).
Here, one unit should correspond to 3 bytes, with each transfer sending 24 bits of data.

ischenz, 2025-06-22:
After setting 
```y += 120;```
 the display stabilized, though slight flickering remains. I suspect the x-coordinate might be the issue. The ST7305 should be configured as 204x200 on a 200x200 screen, but the first 4 columns are probably unused (no actual circuit behind them).
https://github.com/user-attachments/assets/62efc1ca-a21a-4010-84dc-7e1809b00746

olikraus, 2025-06-22:
Right, it should be 0x26 for the 0x2a command. The remaining flicker may also come from the 8Hz configuration, which is activated in the init code.

Looking in your picts, i think the left part (the H from Hello) seems to be an old content from a previous upload. Maybe you should remove power supply for a while and then reconnect and upload the data. 
In other words, I think the display is still shifted too far to the right. Will it be better with this?
``` C++
        u8x8_cad_SendCmd(u8x8, 0x2a);   // column address set
        u8x8_cad_SendArg(u8x8, 0x15);   
        u8x8_cad_SendArg(u8x8, 0x25 );  
```

olikraus, 2025-06-22:
hmmm, ok, regarding the flpping: Does this depend on the delay of the example code:
``` C++
void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.drawFrame(0,0, 200,200);    
    u8g2.drawLine(0,0,23,23);
    u8g2.drawLine(0+24,0,23+24,47);    
    u8g2.setFont(u8g2_font_ncenB10_tr);
    u8g2.drawStr(0,60,"Hello World!");
  } while ( u8g2.nextPage() );
  delay(1000);  // does the flipping depend on this 1s delay?
}
```
I mean will the flipping be longer of you you increase the delay to 4000 or so?

for the offset... the values 0x16 and 0x26 seem to be more correct, right?
```C++
    u8x8_cad_SendCmd(u8x8, 0x2A);   // column address set
    u8x8_cad_SendArg(u8x8, 0x16);   
    u8x8_cad_SendArg(u8x8, 0x26);
```
But you mentioned a four pixel gap above. So for 0x2a, 0x16, 0x26, will there be a four pixel gap on the left?

olikraus, 2025-06-22:
> a 200x200 screen has to be configured as 204x200

hmm true, and because of u8g2, it has to be dividable by 8 also, so the next bigger number, which can used would be 208. Ok, i think one part of the problem is a kind of a memory leak in the u8g2, I need to reconfigure the constructor for width of 208.

olikraus, 2025-06-22:
ok, I did the memory change, which did affect several files, so there is a new release: 2.36.10

If I remember correctly the following updates are noew included:
 
 * 0x2a cmd with  0x16 and 0x26 
 * Internal memory width of 208x200, configured as 200x200 in u8g2 (hope this works...)

Did I miss something?

How will it look? Is the flicker gone? Is the frame now exactly at the border, or is there still the 4 pixel offset?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

ischenz, 2025-06-23:
Thank you very much. It seems you've taken everything into consideration. I’ve updated the code and it’s running successfully. There is indeed a 4-pixel offset on the X-axis, and there also seems to be some offset on the Y-axis.

[image]
https://github.com/user-attachments/assets/dccbe0fd-9152-453a-a207-52cce86e22f3

olikraus, 2025-06-23:
hmm, looking at the picture, I would also think that the mapping table is still a little bit wrong.

Regarding the y offset: Maybe you can update the y+= value a little bit and find a better value.

olikraus, 2025-06-23:
wunderful, i have applied the three changes. Looks really good.

Regarding the flickering: Can you tell me more about your hardware? Which controller? How is the display connected? Is it SPI? 
Maybe the transfer speed is to high, so we could try to use u8g2.setBusClock(1000000); for SPI

Instead of the bus block maybe the post/pre cs wait time is too short:
https://github.com/olikraus/u8g2/blob/bc051bca241ab9a8ee91b58562440f3226eafe74/csrc/u8x8_d_st7305.c#L397-L398

You could try 100ns instead of 20ns

ischenz, 2025-06-23:
In fact, I don't really understand the principle either — I just found this register in the manufacturer's code. Later, I discovered another issue: currently, we have code like `U8X8_CA(0xB9, 0x23)`, but when I checked the datasheet, I found that the lower 4 bits of register B9 don't correspond to any documented function. However, when I set it to 0x20 according to the manual — that is, `U8X8_CA(0xB9, 0x20)` — the display did stop flickering, but it also didn’t seem to work properly. The screen turned into a snowy/static-like pattern.

[image]

ischenz, 2025-06-24:
Thank you, but to achieve the kind of stable, flicker-free display I mentioned above, there are probably still many places that need to be modified, because the array `static const uint8_t u8x8_d_st7305_122x250_init_seq[]` contains too many ST7302-specific configurations.
Here is my configuration, which allows the screen to display properly:
```
static const uint8_t u8x8_d_st7305_122x250_init_seq[] = {
    
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  
  U8X8_C(0x01),                                 // software resets
  U8X8_DLY(100),
  U8X8_C(0x28), 				// display off
  U8X8_CAA(0xC7, 0x26, 0xE9), 			// disable OSC
  U8X8_CA(0xD1, 0x00), 			// Booster disable
  U8X8_DLY(20),
  U8X8_C(0x10),                                 // sleep in: enter sleep mode
  U8X8_DLY(20),
  U8X8_C(0x01),                                 // software reset
  U8X8_DLY(20),                                 // wait
  //U8X8_C(0x38), 				// High Power Mode
  //U8X8_CA(0xEB, 0x02), 			// Disable NVM Load
  //U8X8_CA(0xD7, 0x68), 			// NVM Load Control: Enable ID1 ID2 ID3 Load
  //U8X8_CA(0xD1, 0x01), 			// Booster Enable
  //U8X8_CA(0xC0, 0x80), 			// Gate Voltage Setting VGH=12V (upper 4 bit, 8V-15V); VGL=-5V (lower 4 bit, -5V .. -10V)
  U8X8_CAA(0xD6, 0x17, 0x02),
  U8X8_CA(0xD1, 0x01), 
  U8X8_CAA(0xC0, 0x12, 0x0a),//chen.x
  U8X8_C(0x0C1),                                // Source Voltage Control 1
  U8X8_A4(115,0x3E,0x3C,0x3C), 
  U8X8_C(0x0C2),                                // Source Voltage Control 1
  U8X8_A4(0,0x21,0x23,0x23),       // Source low voltage in reflective and transmissive mode 
  U8X8_C(0x0C4),    //ischen.x
  U8X8_A4(50,0x5C,0x5A,0x5A), 
  U8X8_C(0x0C5),    //ischen.x
  U8X8_A4(50,0x35,0x37,0x37),
  U8X8_CAA(0xD8, 0x80, 0xE9),
  
  //U8X8_CA(0xCB, 0x14), 			// VCOMH: 0x14 = 4V (0x28 = 5V)
  U8X8_CA(0xB2, 0x12),
  U8X8_CAA(0xB3, 0xE5, 0xF6), 			// Update Period Gate EQ Control, why 0x77??? it should be 0x66 according to the datasheet
  U8X8_A8(0x17, 0x77, 0x77, 0x77, 0x77, 0x77, 0x77, 0x71),
  U8X8_CAA(0xB4, 0x05, 0x46), 			// Update Period Gate EQ Control, why 0x77??? it should be 0x66 according to the datasheet
  U8X8_A6(0x77, 0x77, 0x77, 0x77, 0x76, 0x45),
  U8X8_CAAA(0x62, 0x32, 0x03, 0x1f),                                
  U8X8_CA(0xB7, 0x13),
  U8X8_CA(0xB0, 0x32), 			// Duty Cycle... this must be before sleep out, 200x200 display: 0x64 --> 0x32
  
  U8X8_C(0x11),                                 // sleep out: furn off sleep mode
  U8X8_DLY(120),
  U8X8_CA(0xC9, 0x00),
  U8X8_CA(0x36, 0xa4), 			// Memory Control, 0xa4 for the 200x200 display
  
  U8X8_CA(0x3A, 0x11), 			// Data Format 
  U8X8_CA(0xB9, 0x20), 			// Source Setting: Clear RAM off *******<-----
  U8X8_CA(0xB8, 0x29), 			// Panel Setting / Panel Layout 
  U8X8_CAA(0x2A, 0x16, 0x27), 			// col addr
  U8X8_CAA(0x2B, 0x00, 0x63), 			// row addr
  U8X8_CA(0x35, 0x00),
  U8X8_CA(0xD0, 0xFF), 			// Not in datasheet
  //U8X8_CA(0x72, 0x00), 			// Not in datasheet
  //U8X8_CAA(0xB2,1, 5),                  // Frame Rate for High and Low Power Mode (hier: 32Hz and 8Hz) 
  U8X8_C(0x38), 				// enable Low Power Mode...: 8Hz, see above
  U8X8_C(0x29), 				// display on
  U8X8_C(0x20),
  U8X8_CA(0xBB, 0x4F),
  U8X8_DLY(100),

  /*
  U8X8_CAA(0x2A, 0x19, 0x3a), 			// col addr  0x14 < col < 0x3b
  U8X8_CAA(0x2B, 115, 115), 			// row addr (0..250, y in u8g2), 0 is top row in u8g2
  U8X8_C(0x2C), 			// write start
  U8X8_D1(0xff), 			// pixel data
  U8X8_D1(0xff), 			// pixel data
  U8X8_D1(0xff), 			// pixel data
  
  U8X8_D1(0xff), 			// pixel data
  U8X8_D1(0xff), 			// pixel data
  U8X8_D1(0xff), 			// pixel data

  U8X8_D1(0xff), 			// pixel data
  U8X8_D1(0xff), 			// pixel data
  U8X8_D1(0xff), 			// pixel data
  
  U8X8_D1(0xff), 			// pixel data
  U8X8_D1(0xff), 			// pixel data
  U8X8_D1(0xff), 			// pixel data

  U8X8_CAA(0x2A, 0x19, 0x3a), 			// col addr  0x14 < col < 0x3b
  U8X8_CAA(0x2B, 116, 116), 			// row addr (0..250, y in u8g2), 0 is top row in u8g2
  U8X8_C(0x2C), 			// write start
  U8X8_D1(0x80), 			// pixel data
  U8X8_D1(0x00), 			// pixel data
  U8X8_D1(0x00), 			// pixel data
  
  U8X8_D1(0x00), 			// pixel data
  U8X8_D1(0x00), 			// pixel data
  U8X8_D1(0x00), 			// pixel data

  U8X8_D1(0x00), 			// pixel data
  U8X8_D1(0x00), 			// pixel data
  U8X8_D1(0x00), 			// pixel data
  
  U8X8_D1(0x00), 			// pixel data
  U8X8_D1(0x00), 			// pixel data
  U8X8_D1(0x00), 			// pixel data
  */

  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()           			/* end of sequence */
};
```

ischenz, 2025-06-28:
Thank you very much. Next, I will use it, test it, and see if there are any bugs.

ziustc, 2025-09-04:
Thank you @olikraus for seeing my demand and feedback. Actually the LCD manufacturer provides a sample driver which can display correctly on the LCD. I've pushed in my repository. Besides, I pushed ST7305 user manual provided by the manufacturer too.
https://github.com/ziustc/st7305_168x384_LCD_driver.

To be honest in the first glance I thought it was simple that I can place it into u8g2 myself. But unfortunately nothing shown in the LCD. I have to come to you. Hope you have time to look at it and integrate it into u8g2.

Thank you!

[image]

ziustc, 2025-09-28:
Sorry @olikraus I was out for business trip last whole week. Now you are invited again. Please try.

ziustc, 2025-10-04:
@olikraus I think I have solved the problem. I missed some code in function u8x8_d_st7305_168x384. after adding them, and making some modifications related to the new resolution, whole display works correctly now.

latest file is in the repo now.

[image]

olikraus, 2026-01-20:
@bjornlindrooth-svg : I think there are similar questions and it is more related to ESP32 than to the display. Maybe check other issues here with ESP32 and SPI as keyword. In general you  need to use those pins, which are connected to SPI by default. You can also try the 2nd SPI interface. If I remember correctly one user reported to be able to redirect the default HW SPI pins of the SPI subsystem of the ESP32 to some other pins.

olikraus, 2026-03-08:
Thanks for all your contribution.
Special thanks to @VynDragon , the vendor code helped a lot.

emkaygalaxy, 2026-05-09:
i've had success with the following, is this is correct way to do it ?

void setup(void) {
  SPI.begin(11, -1, 12, 40);
  u8g2.begin();
}

based on -

[image]

this code -

#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>

U8G2_ST7305_300X400_F_4W_HW_SPI u8g2(U8G2_R1, 40, 5, 41);

int hours = 10, minutes = 30, seconds = 0;

void setup(void) {
  SPI.begin(11, -1, 12, 40);
  u8g2.begin();
}

void loop(void) {
  seconds++;
  if (seconds >= 60) { seconds = 0; minutes++; }
  if (minutes >= 60) { minutes = 0; hours++; }
  if (hours >= 24) { hours = 0; }

  char timeStr[10];
  sprintf(timeStr, "%02d:%02d:%02d", hours, minutes, seconds);

  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_logisoso32_tr);
  u8g2.drawStr(50, 50, timeStr);
  u8g2.sendBuffer();
  
  //delay(1000); 
}

result -

[image]