olikraus/u8g2 #2663: Where can i get u8g2.c

Tiendo03, 2025-06-16:
I am using STM32F103C8T6 to communicate with LCD 192x64 px using IC KS0108, but i catch a struggle with library, i think it may be fault by lack of u8g2.c, when i complying the program, i catch some faults like 
LCD\LCD.axf: Error: L6218E: Undefined symbol u8g2_cb_r0 (referred from main.o).
LCD\LCD.axf: Error: L6218E: Undefined symbol u8g2_Setup_ks0108_erm19264_f (referred from main.o).
LCD\LCD.axf: Error: L6218E: Undefined symbol u8x8_InitDisplay (referred from main.o).
LCD\LCD.axf: Error: L6218E: Undefined symbol u8x8_SetPowerSave (referred from main.o).
LCD\LCD.axf: Error: L6218E: Undefined symbol u8g2_ClearBuffer (referred from main.o).
LCD\LCD.axf: Error: L6218E: Undefined symbol u8g2_font_ncenB14_tr (referred from main.o).
LCD\LCD.axf: Error: L6218E: Undefined symbol u8g2_SetFont (referred from main.o).
LCD\LCD.axf: Error: L6218E: Undefined symbol u8g2_DrawStr (referred from main.o).
LCD\LCD.axf: Error: L6218E: Undefined symbol u8g2_SendBuffer (referred from main.o).

PS: i has imported folder cscr in u8g2's u8g2-master in the program 
Anyone can help me this problem

Tiendo03, 2025-06-17:
Thank you, i compiled and have no faults more, but can you give me some informations about the function to active LCD 192x64 KS0108, my code: 
u8g2_Setup_ks0108_erm19264_f(
    &u8g2,
    U8G2_R0,
    u8x8_byte_8bit_stm32,          
    u8x8_gpio_and_delay_stm32
);
	u8g2_InitDisplay(&u8g2);
	u8g2_SetPowerSave(&u8g2, 0);
	u8g2_ClearBuffer(&u8g2);
	u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
	u8g2_DrawStr(&u8g2, 0, 0, "HELLO");
	u8g2_SendBuffer(&u8g2);
But when i load my code to STM32F103C8T6 it doesnt show anything, thanks for help too much <3

olikraus, 2025-06-17:
First: Try to increase the y position in your example string. Ref Position for string draw is lower left by default.

Tiendo03, 2025-06-23:
[image]

olikraus, 2025-06-23:
looks like an hardware issue with the display cs lines.

olikraus, 2025-06-23:
The problem could be also, that the byte function doesn't support the CS lines. The original looks like this
https://github.com/olikraus/u8g2/blob/bc051bca241ab9a8ee91b58562440f3226eafe74/csrc/u8x8_byte.c#L379

Maybe you need to call u8x8_byte_set_ks0108_cs() here also (or insert the code behind u8x8_byte_set_ks0108_cs)

Tiendo03, 2025-06-24:
Thank you so much!, I have fixed it, Have a good day