olikraus/u8g2 #2665: ST7588 LCD Latency

bbd666, 2025-06-16:
I'd like to test the latency of the driver.
I'm using an aduino Nano (ATmega328p (OldBootloader)).
My Idea was to switch between "Hello world!" and "Bye World!" with time cycles of 1s. 
Then try to decrease the cycle delay.

Here is my code

```
#include <Arduino.h>
#include <U8g2lib.h>
#include <Wire.h>

const int clk=13;
const int data=11;
const int reset=8;

U8G2_ST7588_JLX12864_1_SW_I2C u8g2(U8G2_R0, clk, data, reset);

void setup(void) {
  u8g2.begin();
}

int i=0;
char* s;

void loop(void) {
  u8g2.firstPage();
  do {
    i++;
    if (i % 2){
    s="Hello World!";
    }
    else
    {
    s="Bye World!";
    }
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(0,24,s);

  } while ( u8g2.nextPage() );
delay(1000);
}
```

But it doesn't work, nothing's changed on the display and the msg is corrupted

bbd666, 2025-06-16:
ok, the answer lies in l210 of the FAQ.
Now the msg can swicth from "bye" to "hello" but it's corrupted from time to time

v1993, 2025-06-16:
FWIW you're trying to draw from an unitialized pointer the first time around, which can't be good. Also, try using hardware I2C and see if situation improves.

v1993, 2025-06-16:
Fixed code (haven't touched the constructor):

```c++
#include <Arduino.h>
#include <U8g2lib.h>
#include <Wire.h>

const int clk=13;
const int data=11;
const int reset=8;

U8G2_ST7588_JLX12864_1_SW_I2C u8g2(U8G2_R0, clk, data, reset);

void setup(void) {
  u8g2.begin();
}

int i = 0;

void loop(void) {
    char* s;

    if (i++ % 2){
        s = "Hello World!";
    } else {
        s = "Bye World!";
    }

    u8g2.firstPage();
    do {
        u8g2.setFont(u8g2_font_ncenB14_tr);
        u8g2.drawStr(0, 24, s);
    } while (u8g2.nextPage());

    //delay(100);
}
```

P.S.: the `_F` constructor doesn't need a page loop and requires `sendBuffer` instead.

olikraus, 2025-06-16:
Using the SW_I2C constructor is also a bad idea for performance, better use the HW_I2C constructor (might require to use the I2C pins of your board)

In fact the inclusion of "Wire.h" is useless for SW_I2C.

I recently did some performance testing on a none-Arduino platform (which additionally allows DMA transfer): https://github.com/olikraus/u8g2/issues/2564
HW variant should be 10x faster then SW emulation of the bus.

Also note, that SPI is usually faster than I2C.

bbd666, 2025-06-16:
As you may know, I'm writing a driver for python and I try to get benefit of your experience with arduinos.
In my python apps, observed that the instruction send_data frequently returns transmission errors. It must be embedded in a try/except loop and repeated until no errors.
Can you tell me about the instructions enable/disable chip which I did n't use until now (I used display on / display off instructions before and after sending data, but it generates lags, that's not satisfactory).

olikraus, 2025-06-17:
You mean the enable/disable chip select wire connection?
This "chip select" (CS) line is only required for SPI, but not for I2C. With SPI, the CS wire is used to reset the SPI communication. It signals the chip that the next clock transition  will transfer bit 0 of the first byte. Without a proper CS the display might be in a kind of a random state (expecting any bit from 0 to 7).

bbd666, 2025-06-18:
Yes I did it, you were right the pointer was not initialized.
I'm using a raspberry because I'm working on a IPradio, I must handle sound, IR-remote control , LCD display ...

bbd666, 2025-06-19:
My real test is to draw a 128x64 bmp pic.
I tried to load it as your example draxbmp example : 

```
#include <Arduino.h>
#include <U8g2lib.h>
#include <Wire.h>

const int clk=13;
const int data=11;
const int reset=8;

#define u8g_logo_width 38
#define u8g_logo_height 24

static unsigned char u8g_logo_bits[] = {0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  
0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  
0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  
0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  
0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  
0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  
0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  
0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  
0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  
0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  
0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  
0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  
0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  
0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  
0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  
0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  
0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  
0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  
0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x00,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  
0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  
[... cut ...]
0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01,  0x01};

U8G2_ST7588_JLX12864_1_SW_I2C u8g2(U8G2_R0, clk, data, reset);

void setup(void) {
  Wire.begin();    
  u8g2.begin();
  u8g2.setBusClock(100000);
}

int i=1;
char* s;

void loop(void) {
  u8g2.firstPage();
    do {
     u8g2.drawXBM( 0, 0, u8g_logo_width, u8g_logo_height, u8g_logo_bits);
  } while ( u8g2.nextPage() );

//delay(100);
}
```

but there is'nt enough mem of course and it's impossible to compile. Sorry, i'm not used to work with arduinos.

bbd666, 2025-06-19:
I think you misunderstood me. Here I'm just relating my experience with an arduino Nano and your driver. When I try to compile and upload the sketch of my previous post, it's just not possible due to unsufficent memory.

>  global variables need 8793 bytes (423%) of dynamic memory. data section exceeds avilable space in board.

I'm not complaining , I try to understand how you designed your driver for arduino in order to write the equivalent for raspberrys. I've already written one, but I'm facing the issues explained in my post ("2 problems etc") which occur with a raspberry and my python driver.

olikraus, 2025-06-22:
> SDL and SDA GPIO's were not input in the constructor

Yes, because the SDL/SDA pins are defined by your hardware. So you probably need to check your board/hardware/Arduino  description where to connect your display. U8g2 will use the Arduino Wire library, which defines those pins.

For official boards the pin numbers are provided here: https://docs.arduino.cc/language-reference/en/functions/communication/wire/

bbd666, 2025-06-24:
I just tested the Hardware I2c mode. 
You were right it's faster, but the screen freezes after a while (< 5s) : 
https://youtu.be/HYxVQjxPMkg

olikraus, 2025-06-24:
> but the screen freezes after a while (< 5s) 

Maybe there is some watchdog active...

bbd666, 2025-06-24:
to my knowledge no, unless it's a default setttings. No freezing with 1000 ms delay.

bbd666, 2025-07-02:
Thks to you, I finally solved all my pbs with the driver ST7588 in python. Now it woks fine for raspberrys, and the issue came from 
....
the pull-up resistors !
PS: maybe you observed that in your driver one page is missing as the LCD is not 128x64 but 132x64