olikraus/u8g2 #2670: Expression: unicode_lookup_table_len == unicode_lookup_table_pos

jie326513988, 2025-07-02:
C:\Users\aabb\Desktop\u8g2>bdfconv.exe -v MiSans-Demibold_my2_U30.bdf -b 0 -f 1 -M GB2312+ft.map -n my_font_30 -o my_font_30.c
Parse File MiSans-Demibold_my2_U30.bdf: 29214 glyph(s) found
Map: exclude=0 from=0/$0 to=255/$ff map=0/$0
Map: exclude=0 from=17207/$4337 to=17207/$4337 map=17207/$4337 (further single glyph logs disabled)
Map: exclude=0 from=65281/$ff01 to=65374/$ff5e map=65281/$ff01
Reduce: Start
Reduce: End
CalculateMaxBBX: x=-2, y=-8, w=45, h=38
CalculateMaxBBX: Encodings x=106, y=331, w=8551, h=192
bf_CalculateMinMaxDWidth: dx_min=0, dx_max=45
bf_CalculateMinMaxDWidth: x_min=-2, x_max=20
bf_CalculateMaxBitFieldSize: bbx.x=6, bbx.y=6, bbx.w=6, bbx.h=5, dwidth=7
RLE Compress: best zero bits 4, one bits 4, total bit size 9094096
RLE Compress: Font code generation, selected glyphs=12596, total glyphs=29214
RLE Compress: ASCII gylphs=196, Unicode glyphs=12400
RLE Compress: Glyphs per unicode lookup table entry=100
RLE Compress: Unicode lookup table len=124, written entries=123
RLE Compress: Unicode lookup table first entry: delta=496, encoding=1068
RLE Compress: Unicode lookup table last entry: delta=7040, encoding=65535
Assertion failed!

Program: C:\Users\32651\Desktop\u8g2\bdfconv.exe
File: bdf_rle.c, Line 898

Expression: unicode_lookup_table_len == unicode_lookup_table_pos

Hello
When generating a large number of fonts, many glyphs will display this error message.

olikraus, 2025-07-08:
Hmm... the size of the internal format is limited. I assume there is a kind of an overflow. 
For further debugging I would need an example .bdf to reproduce the error.

jie326513988, 2025-07-11:
> Hmm... the size of the internal format is limited. I assume there is a kind of an overflow. For further debugging I would need an example .bdf to reproduce the error.

Thank you.

[test.zip](https://github.com/user-attachments/files/21174310/test.zip)

olikraus, 2025-07-18:
The unix version of bdfconv works without problem. I have attached the generated file.
Maybe the .exe is outdated. I might need to create a new version.

[my_font_30.zip](https://github.com/user-attachments/files/21321067/my_font_30.zip)

jie326513988, 2025-07-28:
Thank you, I am looking forward to the new EXE tool.

> The unix version of bdfconv works without problem. I have attached the generated file. Maybe the .exe is outdated. I might need to create a new version.
> 
> [my_font_30.zip](https://github.com/user-attachments/files/21321067/my_font_30.zip)

closed 2025-07-28 by jie326513988