olikraus/u8g2 #2687: 2.08-inch OLED display module with 256x64 resolution, driven by SH1122

StrobesAV, 2025-08-02:
Hi, 
I’m trying to figure out this display connected to ESP32 WROOM in Anduino IDE (example: page_buffer - Direct Access)
https://goldenmorninglcd.com/oled-display-module/2.08-inch-256x64-sh1122-gme25664-65/
VSPI pins configuration:
MOSI: 23
MISO: 19
SCK: 18
SS: 5 ( I think it id Chip Select)
And this is the constructor:
**U8G2_SH1122_256X64_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 5, /* dc=*/ 2, /* reset=*/ 15);** 
SCL from display connected to GPIO 18
SDA - to GPIO 23	
GND to GND
VDC to 3.3V
This enabled:
#define U8G2_16BIT

But I see nothing on the display. What did I miss? Thank you.

olikraus, 2025-08-02:
* Sometimes there is an issue in the test code, did you try the F constructor "U8G2_SH1122_256X64_F_4W_HW_SPI"?
 * U8g2 takes over control on CS line, so maybe don't use ss: 5 but instead use any other GPIO pin
 * For debugging mit might also help to use software emulated SPI with "U8G2_SH1122_256X64_1_4W_SW_SPI"

StrobesAV, 2025-08-02:
Thank You.
 I was about to give up as both top suggestion did not work, but the third produced the Logo!
Hope this will help others with the same hardware:
U8G2_SH1122_256X64_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 4, /* data=*/ 17, /* cs=*/ 0, /* dc=*/ 2, /* reset=*/ 15);        // Enable U8G2_16BIT in u8g2.h