olikraus/u8g2 #2688: ESP32-C2 with SH1106 issues

KASSIMSAMJI, 2025-08-04:
Hey There

I had some issues getting SH1106 to work properly on [ESP32-C2](https://www.aliexpress.com/item/1005006008372573.html?spm=a2g0o.productlist.main.9.59aacibkcibkkX&algo_pvid=0177f7e7-02d4-468f-b7a6-f29e87fc9d6a&algo_exp_id=0177f7e7-02d4-468f-b7a6-f29e87fc9d6a-8&pdp_ext_f=%7B%22order%22%3A%222%22%2C%22eval%22%3A%221%22%7D&pdp_npi=4%40dis%21TZS%218808.87%216599.57%21%21%213.11%212.33%21%40214100f417543352542452344e3539%2112000035591493936%21sea%21TZ%21138452063%21X&curPageLogUid=hQcuScPtWnZy&utparam-url=scene%3Asearch%7Cquery_from%3A) (ESP8684 dev kit) in PlatformIO 

This script works , the hello world text appears persistently on the screen, It's because I had to run sendBuffer() in a while loop
As a result most other functions like drawLine() , drawXBMP(), etc messes up the display when injected in the code

`````````
#include <Wire.h>
#include <U8g2lib.h>
U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/U8X8_PIN_NONE);

void setup() {
 Wire.begin(sda,scl)
 u8g2.setI2CAddress(0x3C * 2);
 u8g2.begin();
}

void loop() {
 u8g2.clearBuffer(); 
 u8g2.drawStr(26, 15, "hello world");
// u8g2.drawLine();  // this renders the whole display useless
 while(1) {
  yield();
  u8g2.sendBuffer();  
  }
}

`````````

But this usual script here works strangely ie: the hello world appears once and disappears in the blink of an eye, the screen remains blank indefinitely

`````````
#include <Wire.h>
#include <U8g2lib.h>
U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/U8X8_PIN_NONE);

void setup() {
 Wire.begin(sda,scl)
 u8g2.setI2CAddress(0x3C * 2);
 u8g2.begin();
}

void loop() {
 u8g2.clearBuffer(); 
 u8g2.drawStr(26, 15, "hello world");
// u8g2.drawLine();  // this works okay here
 u8g2.sendBuffer();  
 delay(1000);
}

`````````

Any help is highly appreciated, Thanks

olikraus, 2025-08-17:
Maybe you need to disable the watch dog timer: esp_task_wdt_deinit()
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/wdts.html

olikraus, 2025-08-17:
For the display problem, maybe you need to reconfigure 0xda hardware register value of the display (see datasheet): Try one of the following (after u8g2 begin):
``` C++
u8g2_SendF(u8g2, "ca", 0xda, 0x00);  
u8g2_SendF(u8g2, "ca", 0xda, 0x10);  
u8g2_SendF(u8g2, "ca", 0xda, 0x20);  
u8g2_SendF(u8g2, "ca", 0xda, 0x30);  
```
Which of the above lines will work for you?

v1993, 2025-08-17:
> Maybe you need to disable the watch dog timer: esp_task_wdt_deinit() [docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/wdts.html](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/wdts.html)

Turning off [`CONFIG_ESP_INT_WDT`](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/kconfig-reference.html#config-esp-int-wdt) is the better way, since that stops it from getting initialized in the first place. IDK if PlatformIO exposes the config, though; the Arduino compatibility layer as a whole is somewhat of a departure from the core esp-idf.