olikraus/u8g2 #2690: hello，SSD1312 、128X32 、 stm3232 、hw i2c，use which constructor ？

Li-Chang-Yi, 2025-08-11:
looks not good，can not showed totally

Li-Chang-Yi, 2025-08-11:
[image]  look like the same

Li-Chang-Yi, 2025-08-11:
here is my main code 

- thread function

[image]

- u8g2 init

[image]

- u8x8_byte_hw_i2c

[image]

- u8x8_gpio_and_delay

[image]

olikraus, 2025-08-17:
hmmm agree, there seem to be duplicated entries for the c-setup with the I2C code: https://github.com/olikraus/u8g2/wiki/u8g2setupc#ssd1312-128x32

olikraus, 2025-08-17:
the line spacing seems to be an issue with the internal 0xda hardware register.

Will any of the following lines solve your problem? Add one of the lines after begin() statement:

``` C++
u8g2_SendF(u8g2,"ca", 0xda, 0x00);
u8g2_SendF(u8g2,"ca", 0xda, 0x10);
u8g2_SendF(u8g2,"ca", 0xda, 0x20);
u8g2_SendF(u8g2,"ca", 0xda, 0x30);
```

https://github.com/olikraus/u8g2/issues/2483

The initial post from here also seems to show the same problem:
https://github.com/olikraus/u8g2/issues/2368

For the SSD1306 it seems to be correct:
https://github.com/olikraus/u8g2/blob/a549a13b13b5fd568111557c1dd7ca3d06fbe21a/csrc/u8x8_d_ssd1306_128x64_noname.c#L61
In fact the SSD1306 requires a "1" at bit pos 1 (0x0?2). Same is true for the SH1106, which also mentions that the POR value is 0x12. Strange, we might need to fix this for the ssd1312.