olikraus/u8g2 #2693: Kan niet uploaden

Pieter204, 2025-08-28:
ik heb verschillende codes gebruik onder andere SSD1306 , SH1106 maar niets ik heb alles zo aangesloten zoals gemeld het werkt niet ik krijg altijd de fout kan niet uploaden ? Wat gaat er Fout ?
#define OLED_MOSI   11 = D1
#define OLED_CLK    13 = DO
#define OLED_DC     9 = DC
#define OLED_CS     10 = CS
#define OLED_RESET  8 = RES

rob040, 2025-09-06:
Please add your comments in English

rob040, 2025-09-06:
Also your previous placed issue #2678 is not up to a minimum standard that someone willing to help could understand what you are saying.

rob040, 2025-09-06:
When I read your comments, it occurs to me that you might be in school talking to your mate next to you... Perhaps you should have placed your comments and frustrations in your project notebook instead of on github.

Pieter204, 2025-09-07:
@rob040 Sorry but my English is not at its best and I do not have a translator.
What I use Arduino Nano, Com11, Processor Atmega328P, Programmer Arduino as ISP.
When I start this PCB NEP1315-51 AY, 7 pins. with this Adafruit_SSD1306 display(/*MOSI*/ 11, /*CLK*/ 13, /*DC*/ 9, /*RESET*/ 8, ​​/*CS*/ 10); U8G2_FOR_ADAFRUIT_GFX u8g2_for_adafruit_gfx;
then it shows nothing, I have also tried others but also nothing, can you tell me which one I can use for this, the other two PCBs work well, I hope you can help me?

Pieter204, 2025-09-07:
Dear me it is difficult for me to translate this into English I will try
to continue with my problem now . I use Arduino Nano , Com 11,
Processor Atmega 328P, Programmer Arduino as ISP , when I start the board NEP 1315 51 AY 7 pins it does not work there is no display it is the smaller model, and the other a little bigger does not work either?
I have 2 other boards with 4 pins Oled01, GM009605 and they do work , why don't the others work ? Can you help me , maybe I need a different driver ?

Pieter204, 2025-09-07:
I get this error message
Sketch uses 8564 bytes (27%) of program storage space. Maximum is 30720 bytes.
Global variables use 635 bytes (31%) of dynamic memory, leaving 1413 bytes for local variables. Maximum is 2048 bytes.
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x64
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x64
avrdude: stk500_recv (
)​​programmer is not responding avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x64 avrdude: stk500_recv(): programmer is not responding avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x64 avrdude: stk500_recv(): programmer is not responding avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x64 avrdude: stk500_recv(): programmer is not responding avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x64 avrdude: stk500_recv(): programmer is not responding avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x64 avrdude: stk500_recv(): programmer is not responding avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x64 Failed uploading: uploading error: exit status 1

rob040, 2025-09-07:
(No need to post everything twice)

How would we know what PCBs NEP1315-51 AY, Oled01, GM009605 are?
Link to datasheet? photo? what info did seller provide?
Please do the footwork, do not let us guess.

Let me guess: the two displays that do work are probably a common 0.96" OLED display module with SSD1306 controller, using I2C interface. The 7 pin variants can usually be configured to I2C, SPI-3 or SPI-4. Just a wild guess...

What is the link to Adafruit GFX library? Why not use the Adafruit GFX library, they generally have a easier learning curve? Try to keep it as simple as possible; don't mix two totally different libraries until you understand what you are doing.

I still do not understand the title "Can't upload" ?

rob040, 2025-09-07:
With your last post I understand the issue title, but it is not at all related to u8g2

Pieter204, 2025-09-07:
this is the link where i bought it
https://nl.aliexpress.com/item/32913603885.html?spm=a2g0o.detail.pcDetailTopMoreOtherSeller.1.94c9XCiBXCiBgu&gps-id=pcDetailTopMoreOtherSeller&scm=1007.40050.354490.0&scm_id=1007.40050.354490.0&scm-url=1007.40050.354490.0&pvid=813ff06d-aeda-4dbf-9fb8-55161ea45d3b&_t=gps-id:pcDetailTopMoreOtherSeller,scm-url:1007.40050.354490.0,pvid:813ff06d-aeda-4dbf-9fb8-55161ea45d3b,tpp_buckets:668%232846%238114%231999&pdp_ext_f=%7B%22order%22%3A%22132%22%2C%22eval%22%3A%221%22%2C%22sceneId%22%3A%2230050%22%7D&pdp_npi=6%40dis%21EUR%213.40%213.40%21%21%213.88%213.88%21%40211b813b17572590739494412ea987%2112000036274472200%21rec%21BE%212088847893%21X%211%210%21n_tag%3A-29919%3Bd%3Acc154e93%3Bm03_new_user%3A-29895&utparam-url=scene%3ApcDetailTopMoreOtherSeller%7Cquery_from%3A%7Cx_object_id%3A32913603885%7C_p_origin_prod%3A

Pieter204, 2025-09-08:
Hello, can you continue with this link? I still can't get the signs to work. I'm at my wits' end.
Can you help me further, perhaps with an example?

Pieter204, 2025-09-19:
Hello, I used the constructor I indicated, and now everything works fine. Thanks for the help.
Until the next problem, see you later.